MarlinFirmware/Marlin #14688: Singlenozzle (cyclops) problem while changing tool

TheMath38, 2019-07-20:
Hi ! 
I've ordered a Y kit (2 in 1 way) for my extruder. I've configured marlin, (SINGLENOZZLE etc) well, everything is correct my setup work as intended.

The issue is that whenever (in a print) i'm changing a tool (T1 or T0), nozzle just stop on the print (wich makes a big blob by the way) and after a few seconds, it resume.

I've made a lot of research about the issue, I found that it was about a M109 command problem. So I removed all the M109 to replace them by M104. Also I removed the M104 T0 S210 (wich is ignored by Marlin) But nothing. It continue always and always. This is going to blow my mind!

If you want, I can transfer my Configuration files (Marlin) and the G-code that i'm trying to run (or maybe I will do that)

Thank for the help! ^^
[M 1.1.9 Anet A8 e3d.zip](https://github.com/MarlinFirmware/Marlin/files/3413820/M.1.1.9.Anet.A8.e3d.zip)
[G_PokerChip-inlays.zip](https://github.com/MarlinFirmware/Marlin/files/3413823/G_PokerChip-inlays.zip)

InsanityAutomation, 2019-07-20:
FYI ive found an issue with the auto temp switching and have it fixed in a local branch. Ill open a PR tonight or tomorrow once I validate on hardware.

Nozzle stopping with M109 is expected behavior btw. The exact function is to wait for the temp to restabalize.

boelle, 2019-07-21:
could we call this a bug? if so i would edit the title and add label

InsanityAutomation, 2019-07-21:
see if #14696 fixes youre issue. Remember this expects a second hotend to be setup, so no tool change script.

TheMath38, 2019-07-22:
Hey ! Thanks for you reply !

So i checked the #14696 and it doesn't seem to fix the problem :/

But I found a temporary solution, I downgraded my Cura version (initially in 4.1.0) to 4.0.0 and no issue at all, the bug is fixed. This is really strange :3 

I'm sending you the G-code from the Cura 4.0.0 version to see if you could fix the issue in the futur update ! ^^
[CFFFP_PokerChip-inlays.zip](https://github.com/MarlinFirmware/Marlin/files/3417807/CFFFP_PokerChip-inlays.zip)

InsanityAutomation, 2019-07-22:
Ill take a look when I get a chance, however 4.1 has proven very buggy in many ways so if going to 4.0 solved the issue, and it wasnt a symptom of the issue fixed here than it sounds like a cura bug, not a Marlin one for sure.

boelle, 2019-09-24:
@TheMath38 still having this problem?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle