MarlinFirmware/Marlin #14689: [BUG] [bugfix-2.0.x] TMC2208 connection on Ramps clone only works on pins 6/11

sstickel, 2019-07-20:
I am having a hard time getting my TMC2208 steppers to work on my RAMPS clone, only one of four steppers is able to communicate.
I am testing this on a MKS Gen L 1.0 with the latest bugfix-2.0.x commit.

I have four TMC2208 for X,Y,Z and Z2. Only Z2 is able to talk to Marlin using the pins 6/11 on the servo connector. All other uarts don't work. (M122: ALL LOW) 

All steppers and cable are fine. If I connect them to 6/11 I can talk to all of them. I've also put my oscilloscope on the tx pin of the X stepper, and there is actual data coming out, so something seems to be wrong with the RX lines.

This does not look like a regression of that bug where only the last stepper is able to communicate. If I switch the pins for X and Z2, then X works.

I have no more ideas what the problem could be, to me this looks like a bug.

In the attached zip file are my config and the pin file.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3413843/config.zip)

teemuatlut, 2019-07-20:
RX pin needs to be interrupt capable.

sstickel, 2019-07-20:
Correct. But the RX pins in the default ramps pin config are interrupt capable. I havent changed them.
https://www.arduino.cc/en/Reference/SoftwareSerial

sstickel, 2019-07-20:
Oh man, that's embarrasing. Just searched for one of the pin numbers that do not work and saw that the Zonestar LCD uses them. Removed that display from the config and now it works on all lines.

Looks like I need a new display. That Anet A8 display is crap anyway.

closed 2019-07-20 by sstickel