MarlinFirmware/Marlin #14691: [BUG] BLTouch Clone Keeps deploying and retract

eviltuna, 2019-07-20:
Just upgrading from an MKS GEN L to an MKS SGEN L and moving from Marlin 1.1.9 to Bugfix 2.0.x, the 3D Touch was working fine in 1.1.9. I copied the settings over to 2.0. My issue is as soon as the printer turns on the 3DTouch continually deploys and retracts the sensor, it even does this when I do G28, G29. Tried using M280 commands and they all seem to work fine.

RudyBenner, 2019-07-20:
uncomment the delay setting, its in the configuration_adv.h file. I think minimum is 200 now.

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

eviltuna, 2019-07-21:
Figured out the problem.

Had to uncomment this line

#define Z_PROBE_SERVO_NR 0 and change it to

#define Z_PROBE_SERVO_NR 1

RudyBenner, 2019-07-21:
Did you try leaving it commented out?
No change from default?

boelle, 2019-07-21:
will close this one then, give a shout if we need to re-open it

closed 2019-07-21 by boelle