MarlinFirmware/Marlin #14695: [BUG] I2C EEPROM conflicts with OLED SSD1309 on Due

berezovskyi-oleksandr, 2019-07-21:
### Description

OLED display on SSD1306 with I2C interface conflicts with I2C EEPROM on my RuRAMPS board with Arduino DUE.

### Steps to Reproduce

1. Enable SSD1306 in Configuration.h by adding `#define U8GLIB_SSD1309`
2. Flash firmwarw
3. Try to initialize EEPROM

**Expected behavior:** Operation succeed

**Actual behavior:** "Error writing to EEPROM" appear in terminal and board reboots

#### Additional Information

I've fond similar problem on arduino forum - https://forum.arduino.cc/index.php?topic=318759 . There was an idea that there is some problems with simultaneous use of Wire and U8Glib libraries.

My current solution was to change `eeprom_init` function at `Marlin/src/HAL/shared/I2cEeprom.cpp` to be not lazy:
```
static void eeprom_init(void) {
  Wire.begin();
}
```
After this change both display and EEPROM works correctly.

boelle, 2019-09-24:
@berezovskyi-oleksandr still having this issue?

berezovskyi-oleksandr, 2019-09-24:
> @berezovskyi-oleksandr still having this issue?

Re-checked on latest commit at [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) branch ([`6fc4d4c`](https://github.com/MarlinFirmware/Marlin/commit/6fc4d4c3b765dbb1095355da7b12b6a1e9a7ad75)). Still have same issue and fix proposed in initial message still works perfectly.

The only difference is that file `Marlin/src/HAL/shared/I2cEeprom.cpp` now at `Marlin/src/HAL/shared/eeprom_i2c.cpp`.

boelle, 2019-09-24:
yep that would be perfect :-D

boelle, 2019-09-24:
remember to put a link or say that is solved this issue

closed 2019-09-27 by berezovskyi-oleksandr