MarlinFirmware/Marlin #14697: [BUG] fysetc_STM32F1.py script don't like the paths with spaces

sl1pkn07, 2019-07-21:
### Description

fysetc_STM32F1.py script for build Cheetah board  don't like the paths with spaces

### Steps to Reproduce

1. download the code from git
2. replace the configuration files from  `config/examples/FYSETC/Cheetah/base/` (also happen with the bltouch variant)
3. edit platformio.ini for set `env_default  = fysetc_STM32F1`
4.  Run `pio run`

**Expected behavior:** [What you expect to happen]

build ok

**Actual behavior:** [What actually happens]

with path with spaces https://pastebin.com/yAUFKQu8

with path without spaces https://pastebin.com/dpqv72pY

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GMagician, 2019-07-21:
OT
Instead of change `env_default` you may write:
`pio run --environment=fysetc_STM32F1`

sl1pkn07, 2019-07-21:
too much text, and nothing change

jmz52, 2019-07-23:
Try to add extra quotes to `"$BUILD_DIR/${PROGNAME}.elf"` and `"$BUILD_DIR/${PROGNAME}.hex"`
This works on Windows 7 except that I have no `tool-stm32duino` package installed.

```
env.AddPostAction(
	"$BUILD_DIR/${PROGNAME}.elf",
	env.VerboseAction(" ".join([
				"$OBJCOPY",
				"-O",
				"ihex",
				'"$BUILD_DIR/${PROGNAME}.elf"',
				'"$BUILD_DIR/${PROGNAME}.hex"'
			]), "Building $TARGET"))
```

sl1pkn07, 2019-07-23:
works!

~~~
arm-none-eabi-objcopy -O ihex "/home/sl1pkn07/aplicaciones/Cheetah/Mar lin/.pioenvs/fysetc_STM32F1/firmware.elf" "/home/sl1pkn07/aplicaciones/Cheetah/Mar lin/.pioenvs/fysetc_STM32F1/firmware.hex"
MethodWrapper(["checkprogsize"], [".pioenvs/fysetc_STM32F1/firmware.elf"])
arm-none-eabi-objcopy -O binary .pioenvs/fysetc_STM32F1/firmware.elf .pioenvs/fysetc_STM32F1/firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  28.0% (used 13760 bytes from 49152 bytes)
PROGRAM: [========  ]  75.5% (used 197924 bytes from 262144 bytes)
.pioenvs/fysetc_STM32F1/firmware.elf  :
section              size        addr
.text              174200   134217728
.text.align             0   134391928
.ARM.exidx            400   134391928
.data                4120   536870912
.rodata             19204   134396448
.bss                 9640   536875032
.debug_aranges      18648           0
.debug_info       3389272           0
.debug_abbrev      296971           0
.debug_line        367267           0
.debug_frame        61588           0
.debug_str         173667           0
.debug_loc         213205           0
.ARM.attributes        41           0
.debug_ranges       41576           0
.comment              126           0
Total             4769925
========= [SUCCESS]
~~~

sl1pkn07, 2019-08-08:
fixed(?) with https://github.com/MarlinFirmware/Marlin/pull/14876

closed 2019-08-08 by sl1pkn07