MarlinFirmware/Marlin #14701: [BUG] No movement possible after PID tune save

HoeckDK, 2019-07-22:
### Description

Just upgraded to SKR v1.3.
Had a succesful print, but temperature was way unstable. PID tune needed.

After successful PID tune and save to EEP => No movement is possible.

### Steps to Reproduce

1st I did the following:
M303 -> M500 did not kill motion but also did not store PID. Did not fix the issue.

Then I found another guide
M303 -> M301 -> M500 seem to have killed motion, but did store PID.

I have seen this before in 1.1.9, but then I could fix with EEP initialize. 
Now I have found no way to restore motion.
Dowloaded the Marlin branch yesterday.

SPI to TMC2130 and TMC5160 should be OK?!

#define X_DRIVER_TYPE  TMC5160
#define Y_DRIVER_TYPE  TMC5160
#define Z_DRIVER_TYPE  TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2130
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC5160

Recv:                    X           Y           Z           Z2          E

Recv: Enabled                        true        true        true        true        false

Recv: Set current        400         400         425         425         300

Recv: RMS current        392         392         397         397         295

Recv: MAX current        553         553         560         560         416

Recv: Run current        11/31       11/31       12/31       12/31       8/31

Recv: Hold current       5/31        5/31        6/31        6/31        4/31

Recv: Global scaler      130/256     130/256                              130/256

Recv: CS actual                      5/31        5/31        6/31        6/31        4/31

Recv: PWM scale          12583109    12583109    255         255         4

Recv: vsense                                      1=.18       1=.18

Recv: stealthChop        false       false       true        true        true

Recv: msteps             16          16          16          16          16

Recv: tstep              max         max         max         max         max

Recv: pwm

Recv: threshold

Recv: [mm/s]

Recv: OT prewarn         false       false       false       false       false

Recv: OT prewarn has

Recv: been triggered     false       false       false       false       false

Recv: off time                       3           3           3           3           3

Recv: blank time         24          24          24          24          24

Recv: hysteresis

Recv: -end               -1          -1          -1          -1          -1

Recv: -start             1           1           1           1           1

Recv: Stallguard thrs    0           0           0           0           0

Recv: DRVSTATUS          X           Y           Z           Z2          E

Recv: stallguard

Recv: sg_result                      0           0           0           0           0

Recv: fsactive

Recv: stst               *           *           *           *           *

Recv: olb

Recv: ola

Recv: s2gb

Recv: s2ga

Recv: otpw

Recv: ot

Recv: Driver registers:

Recv:                    X           0x80:05:00:00

Recv:                    Y           0x80:05:00:00

Recv:                    Z           0x80:06:00:00

Recv:                    Z2          0x80:06:00:00

Recv:                    E           0x80:04:40:00

Recv:

Recv:

Recv: Testing X connection... OK

Recv: Testing Y connection... OK

Recv: Testing Z connection... OK

Recv: Testing Z2 connection... OK

Recv: Testing E connection... OK

EEPROM readout:
Recv:   G21    ; Units in mm (mm)

Recv:   M149 C ; Units in Celsius

Recv:

Recv:   M200 D3.00

Recv:   M200 D0

Recv:   M92 X79.29 Y79.29 Z400.00 E455.86

Recv:   M203 X175.00 Y175.00 Z5.00 E45.00

Recv:   M201 X3000.00 Y3000.00 Z100.00 E10000.00

Recv:   M204 P3000.00 R3000.00 T3000.00

Recv:   M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00

Recv:   M206 X0.00 Y0.00 Z0.00

Recv:   M420 S0

Recv:   M145 S0 H180 B70 F0

Recv:   M145 S1 H240 B110 F0

Recv:   M301 P18.50 I1.30 D64.80

Recv:   M851 Z-1.60

Recv:   M906 X400 Y400 Z425

Recv:   M906 I1 Z425

Recv:   M906 T0 E300

Recv:

Recv:   M569 S1 X Y Z

Recv:   M569 S1 I1

Recv:  Z

Recv:   M569 S1 T0 E

Recv:   M603 L0.00 U100.00

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3416188/Marlin.zip)

**Expected behavior:** [What you expect to happen]
Printer able to move axis

**Actual behavior:** [What actually happens]
Printer not able to move axis

#### Additional Information

I have attached all the files that differ from the branch.

HoeckDK, 2019-07-23:
Is there a way to clear the EEPROM (is it emulated on the SKR 1.3?), and start from scratch? Then I can do some further investigations.

HoeckDK, 2019-07-25:
Just had a look. Now it says mstep 0 🈂️ 

Recv: msteps		0	0	0	0	0
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold
Recv: [mm/s]

tstep max.

ManuelMcLure, 2019-07-25:
If you do ``M502`` do you get things back into a better shape? That should reinitialize your EEPROM with the values compiled into the firmware.

HoeckDK, 2019-07-25:
That changed the readout a bit. And got mstep back to 16. Still no movement

Recv: 		X	Y	Z	Z2	E
Recv: Enabled		true	true	true	true	false
Recv: Set current	400	400	425	425	300
Recv: RMS current	392	392	397	397	295
Recv: MAX current	553	553	560	560	416
Recv: Run current	11/31	11/31	12/31	12/31	8/31
Recv: Hold current	5/31	5/31	6/31	6/31	4/31
Recv: Global scaler	130/256	130/256			130/256
Recv: CS actual		5/31	5/31	12/31	12/31	4/31
Recv: PWM scale	16056570	16056570	255	255	4
Recv: vsense				1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	37	36	max
Recv: pwm
Recv: threshold
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		3	3	3	3	3
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		-1	-1	-1	-1	-1
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: stallguard
Recv: sg_result		0	0	0	0	0
Recv: fsactive
Recv: stst		*	*			*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x80:05:40:00
Recv: 		Y	0x80:05:40:00
Recv: 		Z	0x00:0C:00:00
Recv: 		Z2	0x00:0C:00:00
Recv: 		E	0x80:04:40:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK

HoeckDK, 2019-07-25:
Are M200 supposed to be there twice in the EEP readout?

Recv:   G21    ; Units in mm (mm)
Recv:   M149 C ; Units in Celsius
Recv: 
Recv:   M200 D3.00
Recv:   M200 D0
Recv:   M92 X79.29 Y79.29 Z400.00 E455.86
Recv:   M203 X175.00 Y175.00 Z5.00 E45.00
Recv:   M201 X3000.00 Y3000.00 Z100.00 E10000.00
Recv:   M204 P3000.00 R3000.00 T3000.00
Recv:   M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Recv:   M206 X0.00 Y0.00 Z0.00
Recv:   M420 S0
Recv:   M145 S0 H180 B70 F0
Recv:   M145 S1 H240 B110 F0
Recv:   M301 P18.50 I1.30 D64.80
Recv:   M851 Z-1.60
Recv:   M906 X400 Y400 Z425
Recv:   M906 I1 Z425
Recv:   M906 T0 E300
Recv: 
Recv:   M569 S1 X Y Z
Recv:   M569 S1 I1
Recv:  Z
Recv:   M569 S1 T0 E
Recv:   M603 L0.00 U100.00

ManuelMcLure, 2019-07-25:
You should be able to do an ``M502``, then a ``M301 P18.50 I1.30 D64.80`` to save the PID settins and an ``M500`` to save the combination of the two. At least that way you should have both the motion settings and PID settings.

HoeckDK, 2019-07-25:
Progress... I now got it going again. It took a reboot of the USB powered SKR board while the big power supply was on all the time. Then movement :)

After I had movement once, now I can connect the PSU again and have the SKR turn it on and off, and I do not loose movement. 

Weird!

ManuelMcLure, 2019-07-25:
If the TMC drivers lose power they need to be reinitialized. If you turn on the power supply with ``M80`` and the ``PS_ON`` pin Marlin will do this for you, but the SKR 1.3 doesn't have a ``PS_ON`` pin.

HoeckDK, 2019-07-25:
I have modified the code:
pins_BIGTREE_SKR_V1.3.h
//
// Heaters / Fans
//
#define HEATER_0_PIN       P2_07
#if HOTENDS == 1
  #define FAN1_PIN         P2_04
#else
  #define HEATER_1_PIN     P2_04
#endif
#define FAN_PIN            P2_03
#define HEATER_BED_PIN     P2_05

to 

//
// Heaters / Fans
//
#define HEATER_0_PIN       P2_07
#if HOTENDS == 1
  #define FAN1_PIN         P2_04
#else
  #define HEATER_1_PIN     P2_04
#endif
#ifndef PS_ON_PIN
  #define PS_ON_PIN        P2_03
#endif
//#define FAN_PIN            P2_03
#define HEATER_BED_PIN     P2_05

So I use the FAN pin for the purpose. Working stellar.

HoeckDK, 2019-07-25:
Suggestion for a feature;)

boelle, 2019-09-24:
@HoeckDK so this one is fixed? what do you suggest for a feature (FR)?

HoeckDK, 2019-09-29:
I just had the hickup again now, where the drivers are not moving the mothers. However how I have marlin configured now (PSU default on), I can just reset the SKR board, and it works again. 

I think it would be a nice option to be able to reinit the steppers from G-code.

Also I think it would be nice to have a #define to use the fan connector for PSU control.

boelle, 2019-10-21:
will close this one as it mostly looks like an issue with hardware, ie not correct powerup sequence of logic and drive voltage

we can always reopen if that is not the case

closed 2019-10-21 by boelle