MarlinFirmware/Marlin #14705: [BUG] (short description) Driver error. Printer HALTED Please Reset... LOOP

brainstormi, 2019-07-22:
Hi all...
    One of my TMC2130 drivers got burned meanwhile I was trying to setup SPI in a trigorilla board. The issue up to then was I got communication issue with all the drivers when running M112 for debugging.
    Now I receive the following error just one second after the board boots: 
      Driver error
      Printer HALTED.
      Please reset.
   The major issue is that although the board can be feed by USB changing the jumper, it's not recognized in any way through USB for any computer, so I can't revert back the config to my previous drivers. 
   Is there any way to flash a new firmware without USB access?.
Regards,

boelle, 2019-07-22:
i dont think this is a bug with marlin, have you seen where to ask for support etc?

brainstormi, 2019-07-22:
@boelle I agree maybe this is not a bug in Marlyn itself, indeed more probably could point to some hardware damage in the board due to the incident with the driver. 
   Obviously this board is a chinese one not terrible expensive to replace it, but I was curious if there would be a way to escape from this loop to trace the issue.

gloomyandy, 2019-07-23:
The problem is this. When Marlin decides to halt it is doing so because it has detected a potential problem with the printer. It wants to then put the hardware in as safe a state as possible (so your house does not burn down). It does this by setting the hardware into a safe state and disabling everything (in particular interrupts), this means that USB stack can not run, which means you can't talk to the board. Allowing USB to function in this state (to help you figure out what is going on) would make Marlin a little less safe, which is probably not a good thing, but it is a hard call.

In this particular case you haven't said what board you are using, but many boards (pretty much all of them) have a way for you to update the firmware even if Marlin will not run. If you can do that you can install a version of Marlin that does not define all of the drivers as 2130s and see if that makes the error go away. It will be a slow process, but you can probably work out if the board is usable at all.

brainstormi, 2019-07-23:
The board is a Trigorilla from Anycubic, it's a Ramps 1.4 + Mega2560 combo. I'm not very good at electronics but I see the board has an ISP port so I purchased a Arduino Uno trying to flash a new bootloader/firm using it.

gloomyandy, 2019-07-23:
I'm pretty sure that board will have an external USB to serial converter chip, so really USB access should work in all cases. Sounds like you may have fried the USB chip. My comments above about Marlin not allowing USB to run are really only true for chips that have built in  USB functions which I don't think is the case with your board, but I may be wrong.

OneOfEleven, 2019-07-23:
> One of my TMC2130 drivers got burned meanwhile I was trying to setup SPI in a trigorilla board. The issue up to then was I got communication issue with all the drivers when running M112 for debugging.

M112 is so called 'emergency stop'. The TMC debug command is M122.

But anyway, yes the Trigorilla board has it's own USB comms chip, so whatever is or isn't in the ATmega chip won't make any difference. So re-burning the boot loader into the MCU won't change/fix any USB comm problems.

The best thing you can do to be honest is to remove and replace the Trigorilla board used on the AnyCubic with a nice new BigTreeTech SKR v1.3 32-bit board (very cheap). You can then at least run a delta without the barbones setup, you could enable whatever options you like.

brainstormi, 2019-07-23:
It seems the board has a CP2102 usb serial converter, so your theory make sense. Thank you very much for your help @gloomyandy,@OneOfEleven . This is also a learning exercise for me, maybe I would be able to replace the SMD with a new one... maybe not :) Probably I'll give it a try only for playing a bit and if successful recover the board for future projects.

[image]

I already took a look to that board (BigTreeTech SKR v1.3) yesterday and was near to order it, my concern was more about the compatibility with my current setup with an Anycubic Kossel Plus (Display, end stops, bed level sensor... all stock components in that printer), Do you have any experience with it?. Also not sure about the stability of Marlin 2.0 software with it. I'm currently running 1.1.9.
Running that board on a Delta, does make sense to use in the future with klipper firmware? or is it already plenty powerful on its own to not need this kind of solutions.
Regards,

thisiskeithb, 2019-07-24:
@brainstormi I'd recommend joining the [Anycubic Kossel 3D - SKR Upgrade Facebook group](https://www.facebook.com/groups/1169081219940646/) if you plan to go with the SKR 1.3 in your delta.

tommies1, 2019-07-25:
I dont' know if this could help. 
but in my case with gt2560 after flashing marlin 2.0 i had not access to upload (o usb serial after bootscreen on lcd) . 
to solve my problem had to push reset button each 2  seconds during compilation (before bootscreen disappear on my lcd)  until upload begin. 

May be this could help

boelle, 2019-09-24:
@brainstormi do you still have problems?

brainstormi, 2019-09-24:
@boelle Apologize me for the late notice, I thought I already commented the outcome and closed the issue. First of all thanks everybody for your feedback and support, finally I was able to flash back the firmware using the ISP port, which solved the loop issue with the Driver Error LCD report, restoring board functionality although usb connector continue without working. Root cause as suggested was USB controller CP2102 was fried with the incident. Now I would like to try to replace it, if I can.
Also following your tips I purchased SKR 1.3 board, which looks promising taking in account the price, pending to install it in my printer.

boelle, 2019-09-24:
oki so we can close this one?

brainstormi, 2019-09-24:
Yes, closed.

closed 2019-09-24 by brainstormi