MarlinFirmware/Marlin #14706: [Question] pindebug

GMagician, 2019-07-22:
I'm working to AGCM4 version of this code and I've some issues when analog inputs are read (via extDigitalRead). Cause, I think, is pin reprogrammed from analog to digital by the function.

I was reading into DUE pinsDevug.h and I found:
`#define digitalRead_mod(p) extDigitalRead(p)  // AVR digitalRead disabled PWM before it read the pin`

but `extDigitalRead` is redirected to `digitalRead`. Sure this debugging is working correctly in DUE?

boelle, 2019-07-22:
i think this one is better suited for the discord server

tpruvot, 2019-07-22:
The purpose of this pindebug is more to list the affected pins (and see free ones) than reading analog values

GMagician, 2019-07-23:
M43 does something that stop my analogs reads. But question is about Due and fast 
pwm stopping after m43

GMagician, 2019-07-24:
What is supposed to happens on boards after M43? is it allowed they don't work anymore?
DUE may have fast pwm disabled, AGCM4 ADC disabled see #14718

Bob-the-Kuhn, 2019-07-24:
Yes, analog reads have become a problem.  I haven't figured out how to fix it.

> #define digitalRead_mod(p) extDigitalRead(p) // AVR digitalRead disabled PWM before it read the pin

That is present because some HALs do things differently.  For most HALs the standard DigitalRead is assigned but at least one has a custom function.

closed 2019-07-25 by GMagician