MarlinFirmware/Marlin #14707: [BUG] zebra stipes with tmc drivers on cartesian

IliyaMalecki, 2019-07-22:
### Description
with tmc2208 on skr1.3 im having a lot of zebra stipes, using cartesian kinematics. A friend has that problem too on cartesian, and doesnt have it on coreXY, with exact same settings.

### Steps to Reproduce
im using bigtreetech tmc2208 v3.0, gonna attach the config files. Cant get rid of them lines, so it is rather easy to reproduce with any settings)))

1. all microstepping modes give this result
2. adaptive step smoothing and s-curve doesnt help
3. square wave stepping makes it worse

#### Additional Information
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3419107/config.zip)

[image]

[image]

gloomyandy, 2019-07-23:
See this thread for an investigation into what looks like the same thing on a Prusa-i3-Mk3.

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../stepper-motor-upgrades-to-eliminate-vfa-s-vertical-fine-artifacts/

IliyaMalecki, 2019-07-23:
but tmc 2208 should give much more uniform current than other cheapeys like drv. Other 2208 give way less vfa than mine, so I am clearly doing something wrong with my firmware

gloomyandy, 2019-07-23:
I don't think you are. Read the above thread, it varies from system to system. One thing is that the smoother your rails and other movements are the more you can see the vfa. Other faults can mask them.

IliyaMalecki, 2019-07-23:
I didnt change anything in my kinematics when I changed from cheap drivers to tmc2208, and the problem escalated only with new drivers

IliyaMalecki, 2019-07-23:
ye probably I should buy some of them a4988

DerAndere1, 2019-09-20:
Did you try PID tuning? I heard it helped in another issue that was concerned with wave patterns.

boelle, 2019-10-12:
@IliyaMalecki tried what @DerAndere1 suggested?

IliyaMalecki, 2019-10-12:
@boelle it didnt help much, the problem is not connected with extrusion itself

boelle, 2019-10-21:
have you checked that belts and pulleys are ok? loose screws etc?

are you trying to print faster than the printer can handle?

IliyaMalecki, 2019-10-21:
@boelle everything is checked and working properly. Still having problems

boelle, 2019-10-21:
could you try and attach the model you printed? i will then try and print and see if i get the same result

IliyaMalecki, 2019-10-21:
okay, thank you

boelle, 2019-10-21:
you forgot to attach the model

IliyaMalecki, 2019-10-21:
gonna attach it later, I'm a bit busy right now

boelle, 2019-10-30:
still busy?

boelle, 2019-11-03:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

IliyaMalecki, 2019-11-03:
I'm solving it right now by installing new skr 1.3 and reprinting test models I'm gonna send you. And i fell for the trick, nice job

boelle, 2019-11-03:
could be nice if you also attach the stl file, you might need to make a zip file

or is the test model from thingiverse?

IliyaMalecki, 2019-11-04:
[zebra.zip](https://github.com/MarlinFirmware/Marlin/files/3803771/zebra.zip)

didnt find the stl, i have been using this test for some time, and the original vanished somewhere.

IliyaMalecki, 2019-11-04:
and these are my results
[image]
[image]

boelle, 2019-11-11:
seems to be less that the white print

IliyaMalecki, 2019-11-11:
only thing changed is the board - except visuals of the plastic, of course, white pla shows the problem more

boelle, 2019-11-21:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

closed 2019-11-24 by boelle