MarlinFirmware/Marlin #14710: wave problem 8 / 32 bits...

gauix4, 2019-07-23:
Hello :)
I come back with the problems of vage on the axes
see the pictures
I'm clearly not the only one with this problem I have a lot of printer and most are making waves
I confirm that it does not come from the mechanics because I own twenty different printer so two very large with brushless motor rail and everything has been mounted to the comparator
I work in parallel with Hiwin and I also use their dual loop driver and we always meet its wave problems even with precision mechanics that work normally at micron when it works under Ethercat and at the right temperature
I use many 8/32bits cards they have all this problem

[image]
Config:  SKR 1.3, 24V TMC5160 SPI nema 17 0.9° several tests were done with the options and without option any difference (STEALTHCHOP, SQUARE_WAVE_STEPPING, linear advance Nema 17 0.9°/1.8°....) 

[image]

the room above is the first that I showed you the second I just modify the driver of the extruder by a ST820. This is much cleaner but the waves remain apparent

[image]
Config: Megatronics, 24V,drive Hiwin dual loop, motor Brushless Hiwin, DRV8825 for extrudeur
I tested a lot of configuration and the driver I always had straight waves (TMC2100, TMC2130, TMC2208, TMC5160, DRV8825, LV8729, ST820, Drive loop ...) and several cards ... still the same problem. .. I am open to any proposal to find this problem :)
currently I mount an Ethercat system on the print that print the top pieces to see the result will always be the same

thank you in advance for your reply and thank you for reading so far sorry for my writing I am not English but French so forgive me on the writing thank you :)

gloomyandy, 2019-07-23:
See https://github.com/MarlinFirmware/Marlin/issues/14707 looks very similar to me.

@gauix4 with so many printers/control boards are you using anything other than Marlin on them and if so do you see the same problems? Perhaps worth trying smoothieware or klipper?

gauix4, 2019-07-23:
I tested Lerdge 3D and there was no problem

coreyfaure, 2019-08-06:
I have the same problem on my Wanhao i3 Plus with an SKR 1.3 board, and TMC2208/TMC2209 drivers (tried full sets of both) with the Moon's 0.9 degree steppers mentioned in the article above. I have no other boards to compare this to unfortunately.

DerAndere1, 2019-09-20:
Did you do PID tuning? In a recent issue waves appeared due to the fact that no PID tuning was done. Waves were caused by expansion of the heatbed during heating cycles or something like that.

boelle, 2019-10-12:
@gauix4 tried what @DerAndere1 suggested?

boelle, 2019-10-22:
given that there has been no activity i will close this one

we can always reopen if the problem is still there

closed 2019-10-22 by boelle