MarlinFirmware/Marlin #14711: [BUG] MKS Robin Nano E0_AUTO_FAN_PIN Sanity Check Error

thisiskeithb, 2019-07-23:
### Description
I am compiling a config for the Tevo Nereus (MKS Robin Nano-based w/ "dumb" MKS TFT) with the help of @zillarob which has the hotend attached to HE0 (pin `PC3`), hotend auto fan connected to HE1 (pin `PB0`), and the parts cooling fan connected to the FAN connector (pin `PB1`):

For a general reference: 
[image]

### Steps to Reproduce

Setting `E0_AUTO_FAN_PIN` to `FAN1_PIN` in `Configuration_adv.h` causes the `SanityCheck.h` to return ["You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."](https://pastebin.com/k8rk861e) despite the pins being different for `FAN1_PIN` and `FAN_PIN`:
```c++
//
// Heaters / Fans
//
#define HEATER_0_PIN       PC3
#if HOTENDS == 1
  #define FAN1_PIN         PB0
#else
  #define HEATER_1_PIN     PB0
#endif
#define FAN_PIN            PB1
#define HEATER_BED_PIN     PA0
```

Commenting out the hotend conditional and defining the `FAN1_PIN` directly returns the [same "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."](https://pastebin.com/byUQS4fv) error:
```c++
//
// Heaters / Fans
//
#define HEATER_0_PIN       PC3
//#if HOTENDS == 1
//  #define FAN1_PIN         PB0
//#else
//  #define HEATER_1_PIN     PB0
//#endif
#define FAN1_PIN           PB0
#define FAN_PIN            PB1
#define HEATER_BED_PIN     PA0
```

Taking it a step further and defining `E0_AUTO_FAN_PIN` as `PB0` in `Configuration_adv.h` and commenting out `FAN1_PIN` in the pins file still results in the [same "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."](https://pastebin.com/8qtnb2nC) error:
```c++
//
// Heaters / Fans
//
#define HEATER_0_PIN       PC3
//#if HOTENDS == 1
//  #define FAN1_PIN         PB0
//#else
//  #define HEATER_1_PIN     PB0
//#endif
//#define FAN1_PIN         PB0
#define FAN_PIN            PB1
#define HEATER_BED_PIN     PA0
```

Troubleshooting further with @tpruvot, I modified the pins file again to undefine `E0_AUTO_FAN_PIN` and defined it directly as `PB0`, which still results in the [same "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."](https://pastebin.com/WS7SxMWy) error:
```c++
//
// Heaters / Fans
//
#define HEATER_0_PIN       PC3
//#if HOTENDS == 1
//  #define FAN1_PIN         PB0
//#else
//  #define HEATER_1_PIN     PB0
//#endif
//#define FAN1_PIN           PB0
#undef E0_AUTO_FAN_PIN
#define E0_AUTO_FAN_PIN    PB0
#define FAN_PIN            PB1
#define HEATER_BED_PIN     PA0
```

If you comment out the `#if HAS_AUTO_FAN` block in `SanityCheck.h`, everything compiles without issue and both the auto fan & parts cooling fan works as expected.

Why is sanity check throwing an error when these are clearly two different pins?

Link to a copy of the Tevo Nereus code, ready to compile/test [from my repo](https://github.com/thisiskeithb/Marlin/tree/issue_tevo_nereus) and just in case, here is the Config/pins file: [Tevo Nereus - MKS Robin Nano.zip](https://github.com/MarlinFirmware/Marlin/files/3422733/Tevo.Nereus.-.MKS.Robin.Nano.zip)

thisiskeithb, 2019-07-26:
More troubleshooting from [another issue](https://github.com/MarlinFirmware/Marlin/issues/14732#issuecomment-515569011). 

@tpruvot: Is there a particular file I should look for? There are a ton of files to go through now that I've added the `-save-temps` flag (see screenshot below), but the ones I've checked don't offer any real answers:

[image]

tpruvot, 2019-07-26:
i would check on the file where you got the error in the past

thisiskeithb, 2019-07-26:
Here are all the *.i files if you can point me in the right direction: [tevo-nereus-temp-files.zip](https://github.com/MarlinFirmware/Marlin/files/3437238/tevo-nereus-temp-files.zip).

I'm not seeing an obvious error/reason why `SanityCheck.h` is throwing that that "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN." message.

tpruvot, 2019-07-26:
from board.ii on the VET6 i get that 
# 126 "/root/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103v/board/board.h"
enum {
PA0,PA1,PA2,PA3,PA4,PA5,PA6,PA7,PA8,PA9,PA10,PA11,PA12,PA13,PA14,PA15,
PB0,PB1,PB2,PB3,PB4,PB5,PB6,PB7,PB8,PB9,PB10,PB11,PB12,PB13,PB14,PB15,
PC0,PC1,PC2,PC3,PC4,PC5,PC6,PC7,PC8,PC9,PC10,PC11,PC12,PC13,PC14,PC15,
PD0,PD1,PD2,PD3,PD4,PD5,PD6,PD7,PD8,PD9,PD10,PD11,PD12,PD13,PD14,PD15,
PE0,PE1,PE2,PE3,PE4,PE5,PE6,PE7,PE8,PE9,PE10,PE11,PE12,PE13,PE14,PE15,
};
# 34 "/root/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103v/board.cpp" 2

but the RCT6 is different, with less gpios

boelle, 2019-07-26:
just my 0.02DKK 

do you set the autofan pin in the pins file? i set it in configuration_adv.h

tpruvot, 2019-07-26:
maybe try fgrep -r PB0 * to find conflicting ones...

tpruvot, 2019-07-26:
nope i dont, our nozzle pin is 24V permanent

thisiskeithb, 2019-07-26:
@boelle: See the first post 🙂 I’ve tried setting it multiple ways. The weird part is that if you bypass the sanity check, the board works as expected.

boelle, 2019-07-26:
yeah, i just updated my fork and get the same error if i set auto fan to the same as part cooling fan

i know that some delta printers like the reprap pro fisher used one fan to cool both the cold end and at the same time work as a part cooling fan, https://www.thingiverse.com/thing:1503175

tpruvot, 2019-07-26:
but he doesnt set to the same pin ^^

thisiskeithb, 2019-07-26:
@boelle: I know it can be wired differently/modded, but in order to get a printer config into Marlin, it needs to work on the stock/shipped printers. Telling users who upgrade Marlin to rewire their machine/install mods isn't going to fly.

boelle, 2019-07-26:
nope, excatly my thought, i just took the fisher as an example where i know that auto fan and fan needs to be the same for it to work out the box

thisiskeithb, 2019-07-26:
@tpruvot: [Here are the results](https://pastebin.com/pFhXh4Xd) when searching the temp files for `PB0`.

tpruvot, 2019-07-29:
so i seen some boards have #define ORIG_E0_AUTO_FAN_PIN ...

see Marlin/src/pins/stm32/pins_REMRAM_V1.h may be related

thisiskeithb, 2019-07-31:
I've asked @zillarob to test the latest changes ([Tevo Nereus branch](https://github.com/thisiskeithb/Marlin/tree/test_tevo_nereus)) which now compiles with `E0_AUTO_FAN_PIN FAN1_PIN` set, so we'll see if PR https://github.com/MarlinFirmware/Marlin/pull/14785 fixed the issue and didn't break anything. 🤞

zillarob, 2019-07-31:
I should have some time to give it a run this eve and will let you guys know.

thisiskeithb, 2019-08-01:
PR #14785 fixed the pin enums issue we were all seeing. Thanks @tpruvot! But now the LCD touch buttons no longer work.

Closing this since the originally reported issue is fixed. @zillarob and I will try to work out which commit broke the touch buttons on the stock Robin Nano TFT.

closed 2019-08-01 by thisiskeithb

tpruvot, 2019-08-02:
#14768 i guess, no issue here

zillarob, 2019-08-02:
Got a chance to do some looking and it might be the calibration. I did give some random taps on the new one that didnt work the other day just to check, but didnt go crazy so may have slipped by.

On the mks_touch one I played with for you I had commented out the calibration in config and pasted the mks calibration in the pins file.

On this new version the cal in config is defined, and no mks cal in the pins file. 
The numbers look to be similar, but pretty much inverted +/- between the two cals.

I will be out this weekend so prob wont get to test until sun/mon. Hopefully false alarm =D

zillarob, 2019-08-03:
@tpruvot @thisiskeithb 

Good news, and bad news.

Bad news- The weekend didnt go as planned. There was a big boat and a big rock. The big rock one. Was starting to wonder if I was going to find out how good of a swimmer I am =)

Good news- I got back early and was able to play around. Pasting in the mks calibration got everything working again.

Not sure where the best place to put that in might be, Id guess in the pins file, but I will leave that to sharper minds =D

zillarob, 2019-08-08:
@tpruvot 
Looks like you do a bunch of work on the Longer board?
Seems similar to the nano so I pulled the servo stuff out and pasted into the nano pins and will test BLTouch later this eve.
I was curious if you knew if BLTouch was working on the Longer board?

tpruvot, 2019-08-08:
if you mean the endstop as pwm pin, users need to remove a tiny capacitor to use it.

zillarob, 2019-08-08:
Thanks! 
Looks like the cap is on the end at least so should be easy to push off with the iron.
I dont use the 5th stepper so have PB4 PB5 and PB8 free. They on TIM3 and TIM4 so might give them a shot also.

zillarob, 2019-08-08:
@tpruvot 
Getting closer =D
Seems to be working, but is very jittery. Pin I used was PA11 in the Z- endstop.
The one on there now is a TL clone, but it was working fine on the SKR v1.3 before.
Will have to swap in my genuine BLT and see if it fares better.

zillarob, 2019-08-09:
So I got the iron back out, knocked the 10k resistor R45 off the board, and bridged the pads with solder to see if more sig helped.
 It is rock solid now without a single pin glitch since.

[image]

tpruvot, 2019-08-09:
yep, seems weird a so huge value there...

our cap value was too high for the PWM : 
[image]

zillarob, 2019-08-09:
Ya, I always thought 4.7k was the go-to for these things, but seeing the 10k more and more.

Im trying to see if can use E1 EN, STP, DIR for this also so dont have to mod the board.
I had the pins wrong above so need to fix that, they are actually 
E1 STP PA6 TIM(3,8,1?), E1 DIR PA1 TIM(2,5), E1 EN PA3 TIM(2,5)

I see the open drain stuff in the Longer pins file, but I dont fully understand what that is (yet).
Would these qualify for the OD setup with no pullup like the endstop has, or do I need to keep reading the data sheet =D

tpruvot, 2019-08-09:
the open drain (OD) is when you use a pullup res over the 5V (or more). the chip will only ground it to 0 or be "unconnected" to the chip

zillarob, 2019-08-09:
Thanks for the tip. Looks like I have more homework to do =D
I may need to be really careful with those, they arent 5v tolerant.
Easy enough to pullup to 3.3 and just try it the normal way, but would be nice to not have to add the resistor.

When you set a servo pin, do you know if it enables the internal pullup so we might not have to add anything?

tpruvot, 2019-08-09:
pullup or pulldown are for inputs, servo signal is an output

zillarob, 2019-08-09:
Here is what I have been playing with. Pruned from the Longer file and pins set for the Nano.
Zmin >> servo, Zmax >> Zmin

`//
// PWM
//
#define NUM_SERVOS        1
#define SERVO0_TIMER_NUM  1      // General or Adv. timer to use for the servo PWM (2,5 might be reserved)
#define SERVO0_PIN        PA11   // Pulled up PWM pin (3.3V or 0)`

tpruvot, 2019-08-09:
yep, i said that because you said you removed the Zmin resistor.. which is not the most logic :p

tpruvot, 2019-08-09:
Are you sure it was a 10K one ? those in serial are generally small values "fuses" ones as protection

zillarob, 2019-08-09:
For sure, Im just fumbling my way through this stuff. Have an extra board just in case though =D
Im not certain what that resistor is for, but always guessed it was if somebody hooked up the endstop different or wrong it would keep the current down and protect the chip?

It was 10k on the button with the meter. BTT looks like they use them on the SKR also. Oddly enough the TL Touch worked fine on the SKR.

zillarob, 2019-08-09:
Just tested The BLT on E1 STP PA6 TIM8 and looks good on the Robin Nano.
I did only run it through a couple autolevels and havent printed with it yet.

Originally set up with 10k pullup to 3.3v, but pulled that off just to see what would happen and appeared to work fine without.

ETE-Design, 2019-11-18:
Anyone know if simular option is possible for the MKS Robin Lite? Do I just have to put in the PWM line in the pin definition? Or also remove the 10k on the Z+ for it to work?