MarlinFirmware/Marlin #14713: [BUG] Compilation fails using autoleveling and either command

italocjs, 2019-07-23:
This is more of a "notice" than bug report, 

I was trying to burn my firmware with the version downloaded today, but kept failing, when comparing the old firmware i had (2.0.x but from a few months ago) i found that the code changed from - to 

WORKING
`#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)`

NOT WORKING
`#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)`

Error:
The boundaries are wrong, probe cant reach X, Y and Z, and a lot more

i put the old line with the or ( || )  and it worked, maybe its worth taking a look at this, 

Thanks!

thisiskeithb, 2019-07-23:
What's the actual error message? Can you .zip up your configs and attach them?

I just compiled the latest `bugfix-2.0.x` with a `BLTOUCH` and `AUTO_BED_LEVELING_BILINEAR` with no error messages: [link to my config](https://github.com/thisiskeithb/Marlin/tree/6c700df8b34b6992af2f3510f44f3a4e21190478).

boelle, 2019-09-24:
will close this one as @thisiskeithb did not have an issue

we can alway reopen if @italocjs think the problem is still there

closed 2019-09-24 by boelle