MarlinFirmware/Marlin #14715: [BUG] ABL Probing Behaves as if Center of Bed is 0,0.

AbuMaia, 2019-07-24:
### Description

Trying to run ABL to set up a leveling grid, X & Y home properly, Z homes properly with safe Z homing in the center of the bed, then when bed probing begins, the probe only moves -10x and -10y then starts probing the grid, crashing into the X max end. It does not return the probe completely to 0,0 before probing.

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3424526/Configuration.zip)
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RudyBenner, 2019-07-24:
Line 1065 in configuration.h
Line 249 in configuration_adv.h

reloxx13, 2019-07-24:
sounds stupid, but did u try to turn the printer off, wait 10sec and turn on again?

AbuMaia, 2019-07-25:
Yes, I power-cycled the printer several times, cleared and reloaded the eeprom after every firmware change, power-cycled the octopi. I reverted to a version I had downloaded in March, but now I have an intermittent homing problem where the Y axis moves 6mm away from the endstop switch instead of towards it. X and Z axes seem to be operating properly. Otherwise all is normal, I have a print running now in fact.

AbuMaia, 2019-07-26:
Here is a short video I made demonstrating the problem. I move the X +20mm and back to 0, then the Y +20mm and back to 0, to demonstrate the axes move properly and in the correct directions, it's not a miswired stepper or anything. Then I hit the Home button, and X moves properly, out and back to the switch. But Y only moves away from the switch. I do this twice, then move Y back to the switch 1mm at a time to show Y moves 5mm when homing.

I can move Y manually to the switch and start a print, and it prints just fine, just 5mm offset in where it thinks it is on the bed.

https://youtu.be/WRHOGKPEuQ4

AbuMaia, 2019-08-13:
3 weeks and only one response. Guess I'll have to figure it out on my own.

closed 2019-08-13 by AbuMaia