MarlinFirmware/Marlin #14716: Re-Arm RRDFGSC LCD missing or lagging top half of the display

2kingsrock, 2019-07-24:
### Description

Using the Re-Arm with Ramps 1.6+ connected to RepRap Discount Full Graphic LCD top half display lags or is missing when using the menus

### Steps to Reproduce

1. Scroll through various menus, top half can take additional time to sync with bottom half
2. Prevalent on bed level menu (1 of 9 steps)
3. The top half will be completely missing
* Note. Correctly working on Marlin 1.1.9 with Mega 2560, swapped to 2.x bugfix Nightly on Re-Arm * 

**Expected behavior:** Even if screen lags in menus, it should re-sync 

**Actual behavior:** top half of screen goes missing / lags

#### Additional Information

Screen with Offsets applied is from 2.0 bugfix and is home screen working fine.

[image]
[image]

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-07-24:
So 2.0.x is working perfectly?

remember to add configs as zip

2kingsrock, 2019-07-24:
Updated the original comment.  I can add in video, screen shots and configs later today.  This hardware is  working correctly on a 8bit Mega with Ramps 1.6+ on a printer kit CoreXY running Marlin 1.1.9.  I am using this current CORE XY working printer as a baseline and have been creating a conversion kit and process to migrate from the 8 bit Mega.  The goal is to unplug swap to the RE-ARM to get to 32bit running the 2.0x.  I have a series of different brand screens I can test with also this weekend to see if it is just with the RepRap Full Graphics, or if TFT or other 4 line displays display same artifacts.  looking forward to sorting this minor issue out!

2kingsrock, 2019-07-25:
[backup configs.zip](https://github.com/MarlinFirmware/Marlin/files/3429778/backup.configs.zip)
[image]
[image]

Bed level menu or actions seem to be where this issue is visible / repeatable.  Corner Level or Manual 9pt level.  Bottom half seems to be always correct.

AnHardt, 2019-07-25:
This error is reasoned by changing the display contend inside u8glibs [picture loop](https://github.com/olikraus/u8glib/wiki/tpictureloop).
We do have a [state-machine](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/ultralcd.h#L47-L53) to manage the problems caused by that.
But obviously its use is to complicated for some programmers.

Have a look on working code. See how the states are managed an what they do mean.
I'm tired of that and learned live with the problem.

2kingsrock, 2019-07-25:
1) allocate half local frame buffer (called 'page') in the microcontroller RAM (128x32 pixel)
2) clear page (128x32 pixel), set clipping to lines 0 to 31
3) execute draw()
4) transfer page to the upper half of the display frame buffer.
5) clear page (128x32 pixel), set clipping to lines 32 to 63
6) execute draw()
7) transfer page to the lower half of the display frame buffer

Thank you for the info, helps to rule out hardware conflicts / faults, etc.

boelle, 2019-10-12:
any idea how and if this can be fixed?

boelle, 2019-10-24:
@shitcreek i think this one is an easy to fix ?

boelle, 2019-10-30:
@AnHardt how could we fix this one for good so it does not bite us again?

boelle, 2019-11-03:
@2kingsrock i assume this one is still an issue?

boelle, 2019-11-11:
@AnHardt how can we fix this one for good?

boelle, 2019-11-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-21 by boelle