MarlinFirmware/Marlin #14719: [FR] Enable dual-Z support for bigtreetech SKR v1.3

powertomato, 2019-07-24:
Currently dual-Z is not supported on this board. Most notably because the pins*.h file lacks the propper definitions but there might be other things missing I'm not aware of.

I have added the pin definitions on my branch https://github.com/powertomato/Marlin/tree/skr1.3_dual_z and it works fine with my DRV8825 and TMC2130 drivers. If desired I can create a pull request.

Patag, 2019-07-24:
Hum... I'm currently using this board with dual Z steppers.
When Z_DUAL_STEPPER_DRIVERS is defined in configuration_adv.h, Marlin uses automatically E1 driver. No need to define Z2 pins.

powertomato, 2019-07-24:
@Patag oh I didn't know that. I could swear I had compilation errors without it. But I just checked and it works flawlessly... My bad, I will close the issue.

closed 2019-07-24 by powertomato