MarlinFirmware/Marlin #14720: [BUG] Babystepping causes freeze on delta printer with marlin 2.0.x bugfix but not on older 1.1.x versions - caused by wrong calc with MINIMUM_STEPPER_PULSE and _PULSE_WAIT #13300

Stady234, 2019-07-24:
Continuation from previous thread.. https://github.com/MarlinFirmware/Marlin/issues/13300

Looks like this is still an issue..  I just downloaded the latest commit from today.. 
[image]
Flashed to my delta (Anycubic KLP w/ factory Trigorilla board w/A4988) and left the default "min stepper pulse =1" and sure enough the printer locked up as soon as i turn the lcd knob to any value...

I copied the above outlined replacement lines.. and everything is working as expected.

boelle, 2019-07-26:
sorry for the noop question, but what needs to be changed in what file?

i'm not good a coding but i can edit and submit a PR so this can be fixed

Stady234, 2019-07-26:
https://github.com/MarlinFirmware/Marlin/issues/13300#issuecomment-469597108

vivian-ng, 2019-08-20:
I think this [PR #14028](https://github.com/MarlinFirmware/Marlin/pull/14028) was supposed to fix it, but I can confirm when I try to use babystepping or manual mesh leveling, the firmware will crash and reboot.

boelle, 2019-09-24:
@Stady234 problem solved?

InsanityAutomation, 2019-09-24:
No, still exists after last change targeting it didnt work. Id like to get a delta in hand to play with and trace exactly whats happening. I just dont have one to work with atm.

Stady234, 2019-09-25:
@boelle I have not downloaded and tested a new download in a while as I have been waiting to hear f the updates outlined in the other post that work, had been integrated into the source.

For now, i have not had any issues on my 6 printers making the changes specified.

shitcreek, 2019-10-25:
Confirming issue still exists on MicroMake D1 with last night's build of Marlin 2.0 with and without the suggested fix from #13300 and setting `MINIMUM_STEPPER_PULSE` to 2.

boelle, 2019-11-03:
@shitcreek any news on this one?

boelle, 2019-12-01:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-01 by boelle