MarlinFirmware/Marlin #14721: Exception in thread Thread-1    (trouble with AutoBuild)

mattimm, 2019-07-25:
Hi I'm experiencing a bug while trying to use the AutoBuild to build Marlin 2.0. I've experienced the same bug on vscode, sublime, atom, and the command prompt, so I'm suspecting its an issue with auto_build.py or the version of python I'm running (I'm running 2.7.0, will test others soon). 

I'm seeing an exception being thrown whenever I run "buildroot/share/atom/auto_build.py build".

python version 2.7.0
build_type:   build
starting platformio
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 530, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 483, in run
    self.__target(*self.__args, **self.__kwargs)
  File "buildroot/share/atom/auto_build.py", line 1025, in <lambda>
    self.secondary_thread = threading.Thread(target = lambda q, arg1: q.put(run_PIO(arg1)), args=(que, ''))
  File "buildroot/share/atom/auto_build.py", line 842, in run_PIO
    pio_subprocess = subprocess.Popen(['platformio', 'run', '-e', target_env], stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
  File "C:\Python27\lib\subprocess.py", line 672, in __init__
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 882, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified

The only different thing I've done (other than modifying Configuration.h and Configuration_adv.h for my printer's dimensions) is manually enter "py buildroot/share/atom/auto_build.py build" into the command line because I get an error when it tries to run "python buildroot/share/atom/auto_build.py build" (I changed 'python' to 'py')

boelle, 2019-09-24:
@mattimm is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle