MarlinFirmware/Marlin #14722: power_loss_recovery  z axis rise

niu3niu, 2019-07-25:
### Description

open power_loss_recovery,resume print,but  z axis rise! i guess info.current_position[Z_AXIS] is wrong. 
### Steps to Reproduce

1. when i set a specific value to this function dtostrf(10.0, 1, 3, str_1);then the z raise 10.0mm.
2. i use dtostrf(info.current_position[Z_AXIS], 1, 3, str_1);after resume,The z-axis height is not the height before power-off
3. so my guess is that data stored in BIN file is wrong!!

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

niu3niu, 2019-07-27:
I have solve this problem!!
Step1: uncomment  this macro #define NO_WORKSPACE_OFFSETS,So G92.9 command can work,my machine is Cartesian.
Step2:reference this issue #14230.change saved z axis.
Result: z raised ok!!

camilorodas, 2019-08-03:
buenas tardes, tengo un gran problema es mis impresoras no he podido que cuando haya el fallo por corte el sistema de power loss recovery haga que vuelva a la posicion de Z requerida, uso como nivelación MESH BED LEVELING.

shaggyjoseph, 2019-08-08:
@camilorodas 
u should change your code accodrding to issue#14230.i guess it can help u!!

boelle, 2019-09-24:
@niu3niu is the problem still there?

niu3niu, 2019-09-25:
@boelle  i have solve this problem!thank you  very much!

The-Force, 2019-10-04:
This just happened to me. I wanted to try this feature and enabled it, then I started a print and after a while, I forced a power loss. Then after powering on, I selected the resume print option, it does a xy home with some z rise, heats, goes to print position and just when touching the print, it rises again and starts printing in the air.

niu3niu, 2019-10-11:
@The-Force   upload ur powerloss file

thinkyhead, 2019-10-11:
I'll be pretty busy for the next few days, but I will also be dropping in to fix things here and there. I will try to address this before Monday.

boelle, 2019-10-12:
@The-Force upload ur powerloss file

niu3niu, 2019-10-13:
Have a good day!Freiends!

发自 Android 版 Yahoo 邮箱

The-Force, 2019-10-13:
I will try to upload the file as soon as possible, I'm away from home.

The-Force, 2019-10-15:
I'm back. 
This was the file inside my sdcard, I suppose this is the file you are referring to.
[PLR.zip](https://github.com/MarlinFirmware/Marlin/files/3728079/PLR.zip)

camilorodas, 2019-10-17:
hola mi archivo PLR.cpp es este lo unico que hice de cambios fue:
cuando a impresora carga los datos en Z, cambia el power_loss_min_change a que sume o reste la altura de Z dependiendo del error que encuentre una vez yo recupere la impresion. Antes de hacer esto utilizo el M211 para apagar los finales carrera y que marlin me permite tener coordenadas negativas en el movimiento. Es la unica manera que he encontrado que se arregle esta opcion utilizando como metido de malla MBL. Saludos

[power_loss_recovery.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/3740349/power_loss_recovery.cpp.zip)

camilorodas, 2019-10-17:
Uso como electronica en mis impresoras Arduino mega+ramps 1.6+ LCD 128X64 pix, al tiempo note que cuando se graba el archivo de recuperacion la impresora tenia movimientos no gratos en gemotrias esfericas, debido que en el archivo power_loss_recovery.h renia activadas las opciones:
//#define SAVE_EACH_CMD_MODE
//#define SAVE_INFO_INTERVAL_MS 
Al tener esto activado hacia una especia de retardo debido a la transmisión de datos entre la SD y la board,solucione esto dejando estas lineas comentadas y agregando el power_loss_pin en los pines del final de carrera de Z MAX  en estado alto

boelle, 2019-10-17:
@camilorodas could you translate to english?

camilorodas, 2019-10-17:
hello my PLR.cpp file is this the only thing I made of changes was:
When the printer loads the data in Z, it changes the power_loss_min_change to add or subtract the height of Z depending on the error I encounter once I recover the impression. Before doing this I use the M211 to turn off the race finals and that marlin allows me to have negative coordinates in the movement. It is the only way I have found that this option is fixed using MBL mesh as a method.Finally before starting each impression I use the M420 S1 command in the starGcode so that the mesh saved in each impression can be used, Greetings from Colombia.

power_loss_recovery.cpp.zip

I use electronics on my Arduino mega + ramps 1.6+ LCD 128X64 pix printers, at the same time I notice that when the recovery file was recorded the printer had unpleasant movements in spherical gemotrias, because in the power_loss_recovery.h file the options were activated :
// # define SAVE_EACH_CMD_MODE
// # define SAVE_INFO_INTERVAL_MS
Having this activated towards a kind of delay due to the transmission of data between the SD and the board, the solution to this left these lines commented and adding the power_loss_pin in the pins of the Z MAX limit switch in high state

In my company we offer technical service to different equipment and I have seen that the only printers that do the recovery against off are those that do not have automatic leveling methods, what could this be?

boelle, 2019-10-22:
@thinkyhead did any of the newly attached files give a clue to what is wrong?

camilorodas, 2019-10-22:
@boelle I found the solution in the printers that I use as follows:
When the printer activates the power loss recovery she first recovers the information of the hotel and the hot bed, after this she leaves for the origin to load the coordinates of X, Y, Z. The problem is in this line ..

In order for Marlin to let me move in negative coordinates after the G28 I had to add M211
    gcode.process_subcommands_now_P (PSTR ("M211 S0")); // turn off protection limit switches
    gcode.process_subcommands_now_P (PSTR ("G1 Z0 F200")); // is always in Z0
In order for the printer to go down in Z or up in Z the mm that it needs, I must add the offset that I see in this line of code like this:
gcode.process_subcommands_now_P (PSTR ("G1 Z-0.2 F200"));

So with this I gave an ineffective solution for MBL-level equipment, because these all have different disadvantages

boelle, 2019-10-24:
@camilorodas maybe make a PR so that we can have this fixed faster?

camilorodas, 2019-10-24:
@boelle hello of course that if I just created a float type variable in which the user can enter the offset that he sees in Z after the loss of energy, this option is located in the leveling menu, in the PLR make the change that I told you before, besides that the machine first heats the extruder before doing the G28, I do this to be able to detach the piece from the nozzle when it has been a long time after the loss of energy. Everything is in Spanish is the only inconvenience I see, but the code if I program in English.
[ Marlin ROAL 30 2.0.zip](https://github.com/MarlinFirmware/Marlin/files/3769071/Marlin.ROAL.30.2.0.zip)

I upload the version of marlin that I have modified for my printers because I have made changes to the menu.

thinkyhead, 2019-10-25:
We just patched the handling of Z in Power-Loss Recovery. Please test the latest code to see if the problem is fixed for you. If so, then you can close this issue.

The-Force, 2019-10-28:
It works for me.

boelle, 2019-10-28:
will close this one as fixed

we can always reopen

closed 2019-10-28 by boelle

The-Force, 2019-10-31:
> @The-Force could you share your version of marlin to see that you have different from mine thank you very much

https://github.com/The-Force/Marlin

camilorodas, 2019-10-31:
@The-Force the link you give me does not appear your settings with marlin bugfix 2.0, it appears as when you download it original, what I want to know is how you configured the MESH BED LEVELING so that when your printer recovers it does it correctly

The-Force, 2019-10-31:
@camilorodas I don't have any kind of bed leveling enabled.

camilorodas, 2019-10-31:
@The-Force  friend the recovery without leveling already served without problem, this case was to be able to fix the error of the PLR when I had MBL as a bed calibration method, we need to open the problem again. This still persists. @thinkyhead @boelle

The-Force, 2019-10-31:
@camilorodas why did you ask me then? I never mentioned bed leveling anywere. This issue is not specific for your case only. My case was fixed with the patch and I said so.

The-Force, 2019-11-18:
The other day my printer failed mid-print and then the PLR failed by doubling the height like the old bug. I was using and updated fork so the bug was already fix in theory. 
Ok so then I updated my fork, merged the config and proceeded to test. Now when resuming, the printhead came down into the print, WTF?
Some more testing and I found that setting BLOCK_BUFFER_SIZE back to 16 fixed it. I don't know if other values would fix the problem too. I had it at 128 because I was having problems using the lcd menu while printing and this value fixed it for me. I don't know if this expected and normal, I tell just in case.

venkatkamesh, 2020-05-04:
hmm Solution is simple
i too faced the same issue

open power_loss_recovery.cpp
find this line
`strcpy_P(job_recovery_commands[ind++], PSTR("G92.0 Z0"));    //(If you have 0 by default just ignore else change to 0)                
strcpy_P(job_recovery_commands[ind++], PSTR("G1 Z2"));`

change Z2 to Z0 
 Z2 in sense to raise the Z axis to 2mm so i changed it to 0 and all done..!

Good luck