MarlinFirmware/Marlin #14724: [BUG] [BUGFIX-2.0.X] only one TMC2130 working on SKR 1.3

Dendrowen, 2019-07-25:
### Description

When installing multiple TMC2130 drivers in my SKR 1.3 running on 24V, M122 returns a bad connection to all drivers, 
however:
When I only install one driver, that driver works. When switching the driver with another it still does. Wen switching the driver to another axis, still no problem. When installing 2 drivers simultaneous, both the drivers fail to give a response. 

This tells me that all the drivers work, and all the axis as well.

### Steps to Reproduce

1. Put in 1 driver and test with M122 -> Works
2. Put driver in another axis (e.g. from X to Y) -> Works
3. Change driver with another driver -> Works
4. Put the tested drivers in the tested axes -> Doesn't work.

**Expected behavior:** The drivers all return a valid value on M122

**Actual behavior:** The drivers return 0x00:00:00:00 Bad Response!

#### Additional Information

I included my config files and my slightly adjusted pins file although the original pins file had the same issues.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3432253/Marlin.zip)

Dendrowen, 2019-07-25:
Thought it was working but it still isn't. I'm still getting a bad response although everything seems to work but stealthchop. Even the sensorless homing.

Dendrowen, 2019-07-25:
Another funny detail:
When USE_TMC_SW_SPI is enabled, stealthchop is disabled and when USE_TMC_SW_SPI is enables, stealthchop is disabled. I cannot change it with M569 which makes sense because there is no communication

aivar11, 2019-07-26:
Had similar issue with same configuration.
Please make sure you don't connect pin 3 on TMC2130.
My setup with 5 TMC2130 drivers started working after I snipped off 3rd pin on all drivers.

Pin 3 is marked on this picture with open hardware logo next to Step pin.
https://cdn.shopify.com/s/files/1/0239/9287/products/20160027_2_1024x1024.jpg?v=1551124060

eford321, 2019-07-26:
I'm running 4 TMC2130 drivers on my SKR1.3 with pin 3 in place. 

Did you solder a 2 pin header under the diag pins?
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR-V1.3-TMC2130-SPI-Mode.png](url)
Also, are your drivers set to SPI with the solder bridge? are [CFG4 and CFG5 bridged correctly](https://github.com/bigtreetech/BIGTREETECH-TMC2130-V3.0/blob/master/TMC2130-V3.0RM.pdf)?

Dendrowen, 2019-07-27:
> I'm running 4 TMC2130 drivers on my SKR1.3 with pin 3 in place.
>
> Did you solder a 2 pin header under the diag pins? [...]

I am running the watterott drivers and the SPI and CFG connections are there by default. And yes I am connecting both DIAG pins. 

> Had similar issue with same configuration.
> Please make sure you don't connect pin 3 on TMC2130.
> My setup with 5 TMC2130 drivers started working after I snipped off 3rd pin on all drivers.
>
> Pin 3 is [...]

Are you also running the watterott drivers? If so, I'll try snipping it off.

aivar11, 2019-07-27:
> > Had similar issue with same configuration.
> > Please make sure you don't connect pin 3 on TMC2130.
> > My setup with 5 TMC2130 drivers started working after I snipped off 3rd pin on all drivers.
> > Pin 3 is marked on this picture with open hardware logo next to Step pin.
> > https://cdn.shopify.com/s/files/1/0239/9287/products/20160027_2_1024x1024.jpg?v=1551124060
> 
> Are you also running the watterott drivers? If so, I'll try snipping it off but mine also says DCO on the board on that same pin
> .

I'm running watterott drivers and only modification was to snip off pin 3. My drivers also have DCO marking on the pin. Open source hardware logo was uses on earlier versions.

aivar11, 2019-07-27:
I looked at the schematics of watterott drivers and pin3 on version 1.0 of the driver was disconnected and on v2.0 it is connected to DCO pin. On skr1.3 board pin3 is bridged to pin4 and this pin is part of SPI bus. This configuration will drive SPI bus crazy. 
Bigtreetech itself produces drivers based on watterott driver v1.0 schematics.

Looked also einsy board schematics and there also DCO pin is disconnected.

mikeshub, 2019-07-27:
I had a similar bug with SKR 1.3 and some TMC5160s from Biqu (nice work fellas!). I cut the pin for the clock off the board and jumpered the pin to ground with some magnet wire and it fixed them right up. The problem pin on the schematic is sleep. 

The newer boards for Biqu have fixed this error. However they seem to have about an entire shipping container of the old ones that aren't directly compatible with their own motherboards so caveat emptor.

Dendrowen, 2019-07-27:
Thanks for all the help. I will try this in the morning cause it is night time here in the Netherlands and let you know!

Dendrowen, 2019-07-28:
It worked! Thanks for all the help!

closed 2019-07-28 by Dendrowen

esdkrwl, 2019-10-13:
SKR1.3, Marlin 2.0 and TMC2130 v2 from watterott. There was no way I could get these drivers into SPI mode. Now I just clipped pin 3 on all boards and it just works.

Thank you