MarlinFirmware/Marlin #14726: [BUG] Wait-for-temp without an explicit temp doesn't wait

richfelker, 2019-07-25:
### Description

Issuing a M109 or M190 command with no explicit temperature does not wait for the temperature to reach the target value previously set with M104 or M140.

### Steps to Reproduce

Attempt to print a file that first sets nozzle/bed target temps with M140 and M104, then attempts to wait for both with bare M109 and M190 commands (no S or R argument), then begins printing.

**Expected behavior:** Waiting for the previously-set target temperature to be reached. My expectation is based on the documentation, which admittedly is ambiguous. It has an example bare (no arguments) M109 command, described as "Disable autotemp, wait for temp". I read "disable" and "wait" as two separate imperatives (as opposed to "disable autotemp and disable waiting"), and this agrees with what the stock firmware that came on my printer did before upgrading to Marlin 1.1.9. The idiom that's not working here has also been my go-to solution to CuraEngine issue https://github.com/Ultimaker/CuraEngine/issues/1010 and others seemed to agree it was valid gcode usage.

**Actual behavior:** Print attempts to start immediately with cold hotend and bed.

#### Additional Information

* Printer is Creality Ender 3, using the Marlin-shipped example `Configuration.h` and `Configuration_adv.h` files, except that `LIN_ADVANCE` has been enabled.

richfelker, 2019-07-25:
It seems this was intentional, but I believe it's a functional regression and contrary to documentation: https://github.com/MarlinFirmware/Marlin/commit/ecc537015a

boelle, 2019-07-26:
well if you asked me to wait for a temp but run off without giving me the temp to wait for i would not wait

boelle, 2019-07-26:
@thinkyhead @thisiskeithb @InsanityAutomation would you agree?

richfelker, 2019-07-26:
The documented behavior, as I read it (see explanation above), is that it waits for something, which presumably is the set target temperature if a new target temperature is not included. This was the historical behavior.

If you're unwilling to change it, that's a bit annoying but not a huge problem. But it means I have to go back and get CuraEngine to try to fix this on their end (which probably means submitting my own patch). The "obvious" fix is to just expanding the desired temperature in to my wait command in the start gcode, but Cura only supports expanding config values through the GUI; CuraEngine doesn't do it. So either I have to patch CuraEngine to output the concurrent heating commands right itself (which they seem reluctant to do upstream because of some printers that reboot if you try to heat both bed and nozzle due to weak PSU), patch CuraEngine to support config var expansion in custom start gcode, or add a frontend around CuraEngine that does the same expansions the GUI does.

All of this is why it *was* just easier relying on the historical behavior of the gcode commands, which I figured was intended based on the documentation...

boelle, 2019-07-26:
what do you mean by: you?

richfelker, 2019-07-26:
In "if you're unwilling"? I meant whoever's responsible for making that decision for the project. Sorry for not being clear.

boelle, 2019-07-26:
nobody is direct responsible, if anyone is responsible its all of us.
Everyone can submit a PR with a fix for errors etc

https://github.com/MarlinFirmware/Marlin/commit/ecc537015a

points to 1.1.x if not much mistaken, you should be using 2.0.x as 1.1.x is not updated anymore

lrpirlet, 2019-07-26:
humm…
Gcode M109 name on the reprap site is [M109: Set Extruder Temperature and Wait](https://www.reprap.org/wiki/G-code#M109:_Set_Extruder_Temperature_and_Wait)...
The article then goes on telling that

> M109 in Teacup
Not needed. To mimic Marlin behaviour, use M104 followed by M116. 

I would **NOT call this a BUG but rather a new Funtionality Request** and the title would become something like :  _Please do implement M116_

boelle, 2019-07-26:
if that what @richfelker means then I or he could edit the title to reflect that :-)

7eggert, 2019-08-03:
I cannot see how not waiting on not changing the temperature would be beneficial or the expected behaviour. Also I cannot see how waiting for the bed (only) using M116 would be achieved.3
OTOH, M116 could be hackishly implemented by translating it to M109\nM190 on many printers.

boelle, 2019-10-12:
i will close this one as its not a bug

create a [PR] and explain what feature request you want

closed 2019-10-12 by boelle