MarlinFirmware/Marlin #14728: Linear advance incompatible with S-Curve acceleration

vadsh, 2019-07-25:
As far as I understand current (Marlin 2.0.x) implementation of linear advance uses trapezoidal model incompatible with S-Curve acceleration.

If both (LA and SC) are enabled - underextrusion happens when printer starts to decelerate.

EvilGremlin, 2019-07-25:
I use both for a few months and never noticed that. With jerks and junction deviation. My k-factor is around 0.15, direct extruder.

ManuelMcLure, 2019-07-25:
My understanding is that LA is not perfect when used with S-curve, but it still results in an improvement over not using LA at all. So "no s-curve with LA" is better than "s-curve with LA" which is still better than "s-curve with no LA".

vadsh, 2019-07-25:
> but it still results in an improvement over not using LA at all. So "no s-curve with LA" is better than "s-curve with LA" which is still better than "s-curve with no LA".

Not in my case. I've updated original posts. Please look at the picture.

ManuelMcLure, 2019-07-25:
Did you rerun the K-factor calibration after enabling S-curve acceleration?

vadsh, 2019-07-25:
> Did you rerun the K-factor calibration after enabling S-curve acceleration?

Calibration pattern is another story. I believe it is also subjected to this issue.
I did it multiple times with steps as low as 0.001

The thing is none of the calibration lines seems to be perfect. K=0 is one of the best actually. All the other have different issues. 0.03 is a little better than others (not ideal though).

thisiskeithb, 2019-07-25:
With a proper K value & `S_CURVE_ACCELERATION` enabled, I haven't seen underextrusion like this on Marlin `bugfix-1.1.x` (MKS GEN-L/TMC2208s) or `bugfix-2.0.x` (SKR 1.3/TMC5160s), but I'm printing at 65mm/s or less.

vadsh, 2019-07-26:
> With a proper K value & `S_CURVE_ACCELERATION` enabled, I haven't seen underextrusion like this on Marlin `bugfix-1.1.x` (MKS GEN-L/TMC2208s) or `bugfix-2.0.x` (SKR 1.3/TMC5160s), but I'm printing at 65mm/s or less.

I believe the effect is largely dependent on the path from extruder to meltzone. The tighter it is the more the effect. Bowden extruders should not suffer so much. Lower speeds with high acceleration should also suffer less (due to very low transition time).

I've also added the link to detailed description of this issue from linear advance creator who seems to stop updating it 9 months ago.

thisiskeithb, 2019-07-26:
Both configs are BMG/direct drive, but I haven't pushed speeds like you're running.

vadsh, 2019-07-26:
> I use both for a few months and never noticed that. With jerks and junction deviation. My k-factor is around 0.15, direct extruder.

As 0.15 k-factor is typical for bowden - it seems that you have pretty much space for filament bending between exctuder and meltzone. Which alleviates the symptoms.

I have k-factor of 0.03 due to short capricorn tube between extruder and meltzone.

If you print at lower speeds (around 60 mm) and with acceleration about 1000 - artifacts also should be minimal due to very short and fast acceleration/decelerations periods.

Issue should be most prominent with tight extruder-meltzone space, high speeds and low accelerations.

vadsh, 2019-07-26:
> Both configs are BMG/direct drive, but I haven't pushed speeds like you're running.

what is you k-factor?

ManuelMcLure, 2019-07-26:
I'm also running BMG direct drive into a genuine V6 - I use 0.07 as my K factor for PLA. I use 0 for PETG because I haven't yet run a calibration pattern with PETG.

thisiskeithb, 2019-07-26:
> what is you k-factor?

Depends on the filament. 0.15 is about average.

boelle, 2019-07-26:
me use 0.05 for both PLA and PET-G, no issues as i also use S-Curve and lin. advance

vadsh, 2019-07-26:
Please don't close the issue for a while - I need several days to run tests and make reliable steps to reproduce for others.

tphan26363, 2019-08-05:
Same for me.  K = 0.03 speed = 100mm/s direct extruder.  Sorry for fuzzy picture.  I did notice this after turn on S-Curve. It  did not (or vaguely) show up on short move.
[image]

SFRSKZ, 2019-09-10:
my prints are normal now. but with s curve and junction they were ugly, under or over extruded walls. i also dont use la anymore, i dont print fast and dont need la for quality since prints are already good.  i used marlin 2 (1 week old from now), got annoying sound at y motor while doing certain moves. i guess marlin 2 needs some time to be reliable.

boelle, 2019-09-24:
@vadsh had time to do more testing?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

Crisis64, 2020-04-01:
i have disabled S_Curve & left Lin_Advance enabled

thierryzoller, 2020-04-20:
Do you have the same issue?

Crisis64, 2020-04-20:
Yes I did, but I have disabled S_Curve. I also tested S_Curve, with Junction Deviation, & have reverted all back to classic jerk, with Linear Advance. This seems to produce best prints for me

fiveangle, 2020-04-20:
@thinkyhead - you may want to consider re-opening

thinkyhead, 2020-04-21:
> Yes I did, but I have disabled S_Curve. I also tested S_Curve, with Junction Deviation, & have reverted all back to classic jerk, with Linear Advance. This seems to produce best prints for me

I believe that is the best combination for any printer with a moving (Y) bed, if you want the nice extrusions. S-Curve Acceleration dynamically adjusts the feedrate throughout acceleration and deceleration, and this takes it out of phase with S-Curve Acceleration.

I would hazard a guess that if you are printing at a certain range of speeds (on 32-bit) you could get away with S-Curve Acceleration plus a very small amount of Linear Advance just to clean up corners.

> you may want to consider re-opening

It's not so much "actionable" as "awaiting LA version 2.0." And the expertise and time needed for a rewrite of the planner is not currently available.

I'll put a sanity-check warning into Marlin when these features are enabled together (especially on AVR). At the current time there's no simple way to make Linear Advance work with S-Curve Acceleration. Linear Advance predicts the duration of the move to be able to relax at the right time and with the right amount, and S-Curve Acceleration makes the duration less predictable, especially on longer lines.