MarlinFirmware/Marlin #14731: [BUG] [bugfix-2.0.x] G28 and G29 affected by M420

spilz87, 2019-07-26:
### Description

When I want to print a second time without restarting my printer, the distance between nozzle and bed is no more good.

I use Bilinear bed leveling with BLtouch 
My Z offset is ok and well saved in eeprom 
My first print is alway ok without using babystep or other, just start the printer and start a print from sd card

### Steps to Reproduce

1. Check your z offset and save the value in eeprom

2. Turn off and on your printer

3. Run this gcode : 
> G28
> G29
> M420 S1;Enable compensation using current grid/mesh
> M420 Z1;Gradually reduce compensation until Z=1
> G1 X100 Y100
> G1 Z0

4. The nozzle should be exactly on the bed

5. Run the same gcode a second time (be careful the nozzle can move under the bed)

6. The nozzle is no more exactly on the bed 

**Expected behavior:** [What you expect to happen]
I expect the second G28 disable the auto leveling and that G28 and G29 are not affected by the previous mesh 

**Actual behavior:** [What actually happens]
Running 2 times the same start gcode do not give the same result 

Removing the two M420 lines solves the issue, but I want to configure the Z fade 

Adding M420 S0 before G28 doesn’t solve the issue 

It seems M420 still affect G28 and G29 while it shouldn’t 

Thanks for your help
Tell me if you can reproduire it on your printer

boelle, 2019-07-26:
you forgot to attach your config as zip

@Roxy-3D any clue what could be wrong here?

Aurawin, 2019-08-01:
Bilinear here as well.  Found this issue because my g29 returns astronomical values.
09:23:59.938 : 0      1      2
09:23:59.938 : 0 -3.059 -2.950 -2.874
09:23:59.940 : 1 -2.579 -2.454 -2.549
09:23:59.940 : 2 -2.563 -2.442 -2.575
09:24:00.440 : X:265.00 Y:295.00 Z:10.00 E:0.00 Count A:89600 B:-4800 Z:5964

I grabbed the latest the other day.  First, I had issues with G34 as the bed would not raise to meet the nozzle. I found that increasing Z_PROBE_LOW_POINT to -4 from -2 allowed the nozzle to get close enough.  BTW: Great feature.

Now I have G34 working I can't get a print.  g29's return values are so absurd my nozzle mashes deep into the bed.

Aurawin, 2019-08-01:
> Bilinear here as well. Found this issue because my g29 returns astronomical values.

spilz87, 2019-08-01:
I explain what I found here : [https://github.com/MarlinFirmware/Marlin/issues/9011#issuecomment-517351956](url)
I think it can explain my issue

spilz87, 2019-08-01:
I’m testing it right now

orcinus, 2019-08-04:
I’ve been getting the same issue with MBL, no Z-fade. Every subsequent print gets more and more offset, and i have to apply babystepping for every subsequent print.

spilz87, 2019-08-04:
I’m just having this issue again for subsequent print. 
It seems the babysteps I have to remove are equal to last G29 measure 
But maybe a coincidence

spilz87, 2019-08-04:
after tests, this start gcode seems to be reliable for subsequent prints : 

> M420 Z0; desable Gradually reduce compensation
> M420 S0; desable compensation using current grid/mesh
> M290 P0 ; disabe probe babystep
> M290 Z0 ; reset babystep ?
> M155 S5 ;temperature report
> M27 ;SD print status report
> G21 ;metric values
> G90 ;absolute positioning
> M82 ;set extruder to absolute mode
> M107 ;start with the fan off
> G28 X0 Y0 ;move X/Y to min endstops
> G28 Z0 ;move Z to min endstops
> M117 Nettoyage 1
> G1 Z15
> G1 F5000 Y70 ;StartGcode
> G1 F5000 X216 ;StartGcode
> G1 Z3
> G1 F5000 Y140 ;StartGcode
> G1 F5000 Y100 ;StartGcode
> G1 Z15.0;move the platform up 15mm
> M117 AutoLevel
> G29 ; Auto Bed Leveling
> M420 S1;Enable compensation using current grid/mesh
> M420 Z0; desable Gradually reduce compensation
> ;M0 Remonter Z Probe
> G92 E0 ;zero the extruded length
> M117 Nettoyage 2
> G1 F5000 Y140 ;StartGcode
> G1 F5000 X216 ;StartGcode
> G1 Z3
> G1 F300 E15 ;extrude 15mm of feed stock
> G1 F5000 E14.7 ;extrude -0.3mm of feed stock
> G1 F5000 Y70 ;StartGcode
> G92 E0 ;zero the extruded length again
> G1 Z10
> M77 ;Stop Print Job timer
> M75 ;Start Print Job timer
> M420 Z1; Gradually reduce compensation until Z=1

shitcreek, 2019-09-17:
Cleaning house. LMK if problem still persists and issue needs to be reopened.

closed 2019-09-17 by shitcreek