MarlinFirmware/Marlin #14732: [BUG] (STM32F1 enum port Precompiling issue)

Msq001, 2019-07-26:
### Description

[image]
As shown in the figure, the port definition of STM32F1 is an enum variable. Some precompiled conditions use these enum variables, so can't get the normal results.
for example
 https://github.com/MarlinFirmware/Marlin/blob/ead6bb267dd17b5053426e4129467373e64c164a/Marlin/src/inc/Conditionals_post.h#L780
Z_MIN_PROBE_PIN == Z_MIN_PIN will always true，if cause here 
https://github.com/MarlinFirmware/Marlin/blob/ead6bb267dd17b5053426e4129467373e64c164a/Marlin/src/module/endstops.cpp#L393
when you enable BLTOUCH， all of these endstops reports will be not compiled

Msq001, 2019-07-26:
@tpruvot any ideas

tpruvot, 2019-07-26:
nope, didnt seen such issue here, yet

Msq001, 2019-07-26:
if you enable bltouch.  when you send M119, it will not report X/Y/Z endstop status

thisiskeithb, 2019-07-26:
I wonder if this bug is related to the issue I'm seeing over in PR https://github.com/MarlinFirmware/Marlin/issues/14711 (also an STM32F1 board).

tpruvot, 2019-07-26:
i would try https://stackoverflow.com/questions/9041381/gcc-preprocessor-output-and-compilation-in-one-pass to get more data... something looks weird in your 2 issues

Msq001, 2019-07-29:
@thisiskeithb Yeah, that's the same reason. 
for example
> #include <stdio.h>
> #include <stdlib.h>
> 
> enum t
> {
>    a,
>    b,
> };
> 
> int main()
> {
>     printf("b = %d\n", b);
>     #if (b==0)
>         printf("b=0\n");
>     #elif (b==1)
>         printf("b=1\n");
>     #endif
>     return 0;
> }

The result of this program is

> b = 1
> b=0

tpruvot, 2019-07-31:
should be fixed with PR #14785 (merged)

thisiskeithb, 2019-07-31:
I'm getting a bunch of new error messages, but I'll work that in a new issue.

tpruvot, 2019-07-31:
here, tested ok on the vet6 and rct6, what kind of new issues ?

thisiskeithb, 2019-07-31:
Warnings about the custom bootscreen (It’s not enabled) and some others. I’m going to start with a clean copy and port settings over to start fresh later tonight.

closed 2019-08-06 by Msq001