MarlinFirmware/Marlin #14734: [FR] Support for G10 L20 P1-6 X0 Y0 Z0 work coordinates (G54-G59)

telleropnul, 2019-07-26:
Marlin is more and more used for CNC machines i.e. MPCNC.
CNC.js is a popular gcode sender.
CNC.js supports work machine / work (offset) coordinate systems.
CNC.js sends a G10 L20 P1 X0 Y0 Z0 command to zero G54 work (offset) coordinates.

Marlin already supports G53 machine coordinates and G54-G59 work (offset) coordinates but unfortunately does not support G10 L20 P1 X0 Y0 Z0.

A more detailed explanation:
https://github.com/synthetos/g2/wiki/Gcode-Coordinate-Offsets

Discussion on the MPCNC forum:
https://www.v1engineering.com/forum/topic/marlin-versus-work-coordinates-g10-l20/

An extremely detailed explanation:
https://coeleveld.com/mpcnc-part-6/

fantasmag0ria, 2020-01-01:
Hi, @telleropnul is it kinda what you had in mind?

https://github.com/cncjs/cncjs/issues/549

telleropnul, 2020-01-08:
Thank you very much for looking into this.

This ticket is now closed and progress is tracked in cncjs/cncjs#549

closed 2020-01-08 by telleropnul