MarlinFirmware/Marlin #14737: Marlin 2 Error Z_MIN_ENDSTOP_INVERTING

ice2009, 2019-07-26:
Hello :) 

I using capacity sensor for my printer (NPN). In my configuration.h i set
 #define Z_MIN_ENDSTOP_INVERTING **true** 
but i need 
#define Z_MIN_PROBE_ENDSTOP_INVERTING **false** 

bacause my sensor is npn. I connect sensor to Zmin on my board. But when i compiling this file i have error 
[image] this configuration works with me in marlin 1.1.9 but i use need marlin 2.0 because i use 32bit board with tmc2209. Anyone could you help me?

ManuelMcLure, 2019-07-26:
Did you actually read the error message? If you’re using both a Z end stop and a Z probe (I.e. ``Z_MIN_PROBE_ENDSTOP_USES_Z_MIN_ENDSTOP_PIN`` is unset) then you must set ``Z_MIN_PROBE_PIN`` to tell Marlin where your probe is connected. If you don’t have a separate Z end stop and are using the probe for homing then ``Z_MIN_PROBE_ENDSTOP_USES_Z_MIN_ENDSTOP_PIN`` should be enabled.

ice2009, 2019-07-27:
Okay but i connect sensor to zmin on board. I don't know pin number. I use fystec cheetah board.

boelle, 2019-07-27:
where do you connect Z endstop? as @ManuelMcLure  said you are using both and they cant be connected to same pin

you also forgot to read the template when you created the issue, it tells you to attach your 2 config files as a zip file

ice2009, 2019-07-27:
I connect my npn sensor to z_min on fystec cheetah board (see on image). I need Z_MIN_ENDSTOP_INVERTING to true function but Z_MIN_PROBE_ENDSTOP_INVERTING must be false because when i use true npn is still trigger. 
[image]

boelle, 2019-07-27:
where do you connect Z endstop (not sensor/probe)?

attach 2 config files as zip

ice2009, 2019-07-27:
> where do you connect Z endstop (not sensor/probe)?
> 
> attach 2 config files as zip

I replace z endstop (ender3) to ljc18a3-h-z/bx.

I use this configuration https://github.com/esamwis3/cheetah-bltouch end i uncomment bltouch and change Z_MIN_ENDSTOP_INVERTING  to true and Z_MIN_PROBE_ENDSTOP_INVERTING to false

boelle, 2019-07-27:
you cant just use a config from outside marlin, you have to copy over settings manual

you can use a tool like winmerge to compare 2 files and make it easy to compare and copy over settings

boelle, 2019-07-27:
here are the Ender 3 example configs 

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Creality/Ender-3

ice2009, 2019-07-27:
> here are the Ender 3 example configs
> 
> https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Creality/Ender-3

Yes i use all marlin https://github.com/FYSETC/Marlin-2.0.x-FYSETC and configuration for ender3 but when i replace z_endstop on marlin to npn capacity sensor and use Z_MIN_ENDSTOP_INVERTING true and npn endstop false i have error from first image.

boelle, 2019-07-27:
that is not marlin, its a copy/fork of marlin and that is a different thing

use https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

and start out with the example config: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Creality/Ender-3

shitcreek, 2019-09-17:
Cleaning house. LMK if problem still persists after using the official Marlin copy and issue needs to be reopened.

closed 2019-09-17 by shitcreek