MarlinFirmware/Marlin #14739: [BUG] Marlin downloads with Files Missing.

Osprey19, 2019-07-27:
### Description

every time i download marlin, there are two files missing from the zip. .pioenvs and .piolibdeps.

### Steps to Reproduce

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information
[image]

thisiskeithb, 2019-07-27:
Those won't show up until you open the project in your IDE/compile Marlin.

teemuatlut, 2019-07-27:
`.piolibdeps` is the folder where PlatformIO automatically downloads dependencies like external libraries.
`.pioenvs` is the folder where PlatformIO builds the Marlin project and the final binary.

Don't forget to close the issue if you have no further questions =)

boelle, 2019-07-27:
i will close this one as its clearly not a bug, if you have further questions have a look here:

https://github.com/MarlinFirmware/Marlin/issues/14694

a list of places where you should ask questions

closed 2019-07-27 by boelle