MarlinFirmware/Marlin #14740: correctly compile MKS Robin,but download to board,it is not work.

niu3niu, 2019-07-27:
### Description

I have correctly compile MKS Robin,but use firmware to update board,but it is not work! 

### Steps to Reproduce

1. [First Step]
I have write a bootloder  for MKS Robin.I know it has change  flash address according to mks_robin.py file.Its flash address is 0x08007000.so in my bootloder,i have relocated flash to 0x08007000. 
2. [Second Step]
Then i copy firmware.bin to sd card.but not response.
3. [and so on...]
i know robin.bin is encryption.so i guess their bootloder must have some code that encode robbin.bin.but i just copy firmware.bin.It is not encrpte.
4.
when i compile robin's firmware project.i update bin file using my bootloder,it can work !!

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-07-27:
1) Where do you download marlin from?
2) Attach a ZIP file containing your Configuration.h and Configuration_adv.h files.

3Dangel, 2019-07-28:
The .bin file has to be Robin.bin
There is files that you need to check before compiling :
- what board do you have ( TFT)
- there is a calibration tool that allows you to see what TFT controller is in. If different than the supported one, well, the Marlin will not work. 

I have Mks-Robin V 2.1 with Touch TFT V 1.0

I spent 6 hours yesterday to have :
- Marlin V2 fix bug  installed
- LCD responsive
- NO SD card

Since my board has WIFI module included on the board, I think it is taking up the serial ports and not allowing the SD card’s to be seen. 

I would need help to understand or to know how to add serial ports through out the Marlin ecosystem
I tried  adding “# define serialport”
And changing to  4,5,6
It did not help. 
So, I’m stuck. 
( I have added lines that “should have helped to activate SD cards but they did not help)
Any help?
Thanks

3Dangel, 2019-07-28:
Well I guess because today is my BD ( for real), spent another 5 hours and try on error ( I’m not a programmer) I got my SD card to work. 
It has to initialized manually, but at least I now can read my STL file from the SD card. 
I would love to attach screenshoot, but I don’t know how. 
It had to do with the Mks-pin out.h

The .h file is not complete. 
I added all pins that I know off and -1 the SD card detect pin. 
Hope this encourage you. 

Cheers

niu3niu, 2019-07-29:
Thanks,Today I try again.I don't know why it can work suddenly.
My lcd is REPRAP_DISCOUNT_SMART_CONTROLLER

niu3niu, 2019-07-29:
> 
> 
>     1. Where do you download marlin from?
> 
>     2. Attach a ZIP file containing your Configuration.h and Configuration_adv.h files.

Thanks.It have work now.

closed 2019-07-29 by boelle