MarlinFirmware/Marlin #14741: [BUG] pid bed temp causes printer halt SKR 1.3

cinealfa, 2019-07-27:
pid bed temp causes printer halt

Printer throws a error and halts bed heating when pidtemp is selcted

### Steps to Reproduce

1. select pid bed temp in config
2. compile
3. watch the fun begin...

**Expected behavior:** [What you expect to happen]
pid control of the bed

**Actual behavior:** [What actually happens]
printer is halted
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3437864/Archive.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-07-27:
Your configs are a bit dated, so try pulling down a fresh copy of [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) and see if the bug persists. I was able to run a bed PID autotune on my SKR 1.3 successfully running a current version.

cinealfa, 2019-07-27:
ah will do! I have been using marlin config tool to set them. I suspected it wasn't updating the build reliably.

mikeshub, 2019-07-27:
When you enabled it did you remember to load the default settings? That happened to me early today and the gains loaded as zero. M502 then M500 should sort you out or use M304 then M500. Be warned the current state of the autotuner might result in an over tuned bed that has temp oscillations. YMMV. Check this PR for more info:

https://github.com/MarlinFirmware/Marlin/pull/14746

cinealfa, 2019-07-27:
it was a older version of marlin apparently the act is no longer being updated so the numbers bugs in it are starting to make it unusable. There hasn't been any work done on it for some time and i fear it has been orphaned.

closed 2019-07-27 by cinealfa