MarlinFirmware/Marlin #14742: [BUG] No Display RemRam V1

leseaw, 2019-07-27:
### Description

Building with graphic display is no longer possible

### Steps to Reproduce

1. use RemRam V1 Board
2. #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
3. Build

**Expected behavior:** Build without Errors and with Display, my last Marln V2.0 Version from january is working, but the file HAL_LCD_com_defines.h has been heavily modified now.

**Actual behavior:** Build with errors and no Display

#### Additional Information

.zip and error messages
[marlin-v2.zip](https://github.com/MarlinFirmware/Marlin/files/3437994/marlin-v2.zip)

boelle, 2019-07-27:
look in your errors: Multiple libraries were found for "U8glib.h"

leseaw, 2019-07-27:
No its not the problem, the problem is in marlin 
With januray version:
look january.txt
[january.txt](https://github.com/MarlinFirmware/Marlin/files/3438000/january.txt)

with version from yesterday:
look july.txt
[july.txt](https://github.com/MarlinFirmware/Marlin/files/3437999/july.txt)

boelle, 2019-07-27:
why do you have 2 folders with the library U8glib ?

boelle, 2019-07-27:
that is one of the errors, you cant have multiple folders with the same library

at least its easy to get that error to go away

leseaw, 2019-07-27:
was an test, new version or not.
Build all on the same machine.
Its the same version in both folder, U8glib at version 1.19.1
look in the HAL_LCD_com_defines.h
is completely new
have here the file from january
[jan-HAL_LCD_com_defines.zip](https://github.com/MarlinFirmware/Marlin/files/3438007/jan-HAL_LCD_com_defines.zip)

leseaw, 2019-07-27:
> 
> 
> that is one of the errors, you cant have multiple folders with the same library
> 
> at least its easy to get that error to go away

no, i build before without the second lib, the same.

boelle, 2019-07-27:
its a strange one, personally i also use the same display and have no problems

leseaw, 2019-07-27:
on the same board? RemRam V1?

boelle, 2019-07-27:
no, i have the re-arm board

leseaw, 2019-07-27:
its not the same :) other chip 32bit STM32F765VI 
Re-Arm NXP LPC1768

boelle, 2019-07-27:
that must partly explain things, @thinkyhead who is it that does STM32 stuff?

leseaw, 2019-07-27:
#14669 this is an problem
Version from 29.06.2019 is working without errors

boelle, 2019-07-27:
so maybe @tpruvot know what is going on here?

leseaw, 2019-07-27:
Version Marlin V2.0 20.07.2019 isnt working :(

tpruvot, 2019-07-27:
Sorry ive no experience for now with the STM32F7

Ludy87, 2019-07-28:
> ### Description
> Building with graphic display is no longer possible
>
> ### Steps to Reproduce
> 1. use RemRam V1 Board
> 2. #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> 3. Build
>
> **Expected [...]

Die `stm32 1.6.1` hat wohl einen Bug kehre zu `\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.0`  zurück

leseaw, 2019-07-28:
@Ludy87 no this is not the problem.
I can build without display.
Test with V1.5, V1.6 and V1.6.1
every time the same error messages

leseaw, 2019-07-28:
@tpruvot @ktand
change in HAL_LCD_com_defines.h
#elif defined(ARDUINO_ARCH_STM32)
    uint8_t u8g_com_arduino_std_sw_spi_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr);
    #define U8G_COM_HAL_SW_SPI_FN u8g_com_arduino_std_sw_spi_fn
    uint8_t u8g_com_stm32duino_hw_spi_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr);
    #define U8G_COM_HAL_HW_SPI_FN u8g_com_stm32duino_hw_spi_fn
    uint8_t u8g_com_arduino_st7920_spi_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr);
    #define U8G_COM_ST7920_HAL_SW_SPI u8g_com_arduino_st7920_spi_fn
	uint8_t u8g_com_arduino_st7920_hw_spi_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr);
    #define U8G_COM_ST7920_HAL_HW_SPI u8g_com_arduino_st7920_hw_spi_fn

These changes are necessary for STM32 HAL
A change in b6546ea33a0f6eebee520dda516e04d3b68ded55 broke the build.

tpruvot, 2019-07-29:
could you share your Marlin/Configuration.h and _adv.h to make a test case with a LCD for the F7 ?

leseaw, 2019-07-29:
Don't you need
Lasts to put RemRam in the place of the board
And full graphic display to choose from.
Or you take the .zip from the first post :)

thinkyhead, 2019-08-08:
> may need a paste of your platformio.ini entry... because the STM32F7 is broken:
> Error: Unknown board ID 'disco_f765vg'

I've just been poking through the `platform-ststm32` looking for something to replace this. So far nothing obvious. Maybe some other board will at least compile, which would be a good start.

thinkyhead, 2019-08-08:
`remram_v1` … ok, well that works!

tpruvot, 2019-08-08:
the closest one seems the 769, i was digging history and this "disco 765" board is not a STM32 discovery one, but a "Borg" https://github.com/Spawn32/STM32GENERIC/tree/master/STM32/variants

thinkyhead, 2019-08-08:
I just set `board = remram_v1` for the Travis CI test, and that is working.

thinkyhead, 2019-08-16:
Applied changes suggested by @leseaw above.

leseaw, 2019-08-28:
I'll be able to test it tonight or tomorrow.
if it goes I close here.

boelle, 2019-09-24:
@leseaw still having this issue?

leseaw, 2019-09-24:
ohh sorry :/

closed 2019-09-24 by leseaw