MarlinFirmware/Marlin #14743: [BUG] G54 + G92 X0 Y0 inadvertently resets origin in G53 machine coordinate system as well.

telleropnul, 2019-07-27:
You can run this on any Marlin gcode parser to replicate:

<power cycle device>
G21 ; millimeters
G90 ; absolute coords
G53 ; machine coordinate system
M114 ; get position
    X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
G54 ; switch to work coordinate system
G0 X10 Y10 ; move
    X:10.00 Y:10.00 Z:0.00 E:0.00 Count X:2000 Y:2000 Z:0
G92 X0 Y0 ; set current position as new origin (similar to G10 L20 P1 X0 Y0)
M114 ; get position
    X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:2000 Y:2000 Z:0
G53 ; switch back to machine coordinates
M114 ; get position
**X:0.00 Y:0.00** Z:0.00 E:0.00 Count X:2000 Y:2000 Z:0

Setting an origin in G54 inadvertently also sets it as origin in G53.  G53 should not be affected and should display absolute machine coordinates at all times i.e. X:10.00 Y:10,00 at this point.

If we repeat this using G55 we get the following:
<power cycle device>
G21 ; millimeters
G90 ; absolute coords
G53 ; machine coordinate system
M114 ; get position
    X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
**G55** ; switch to work coordinate system
G0 X10 Y10 ; move
    X:10.00 Y:10.00 Z:0.00 E:0.00 Count X:2000 Y:2000 Z:0
G92 X0 Y0 ; set current position as new origin (similar to G10 L20 **P2** X0 Y0)
M114 ; get position
    X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:2000 Y:2000 Z:0
G53 ; switch back to machine coordinates
M114 ; get position
**X:10.00 Y:10.00** Z:0.00 E:0.00 Count X:2000 Y:2000 Z:0

Note how Machine Coordinates are not affected by zeroing X and Y in G55 using G92 X0 Y0 and continue to display the true absolute machine coordinates at all times.

shitcreek, 2019-08-08:
I believe there is a bit of confusion here. G53 is not a workspace itself, it is a modifier. It tells the printer to move in accordance to the selected workspace ie G54 - 59.3 : http://marlinfw.org/docs/gcode/G054-059.html

```
G21
G90
M114 ; X:4 Y:2 Z:0
G0 X10 Y10
G55 ; Select workspace 1 | X:10 Y10
G92 X0 Y0 ; X:0 Y:0
G28 XY ; X:4 Y2
G53 G0 X10 Y10 ;  X10 Y10 - but is actually X20 Y20
G54 ; Select workspace 0
M114 ; X20 Y20
```

G53 can be used like so:
G53
G0 X# Y#
--- or ----
G53 G0 X# Y#

telleropnul, 2019-08-09:
"G53 - Move in Machine Coordinates"
http://marlinfw.org/docs/gcode/G053.html

G53 = Move in machine coordinates.
G54 = Move in work offset coordinates.

Here is a screenshot of CNC.js to clarify how a CNC machine operates:

[image]

G53 will always show true absolute coordinates.  When using endstops, you will only be able to move in the positive X and positive Y quadrant.

If you want to center your spindle on a work piece, you can switch to G54 and zero X and Y to create an origin.  You will now be able to move in all four quadrants i.e. negative X and negative Y as well.

As a CNC machinist I am not confused.  Marlin currently does not follow industry standard G53 and G54 conventions.

telleropnul, 2019-08-09:
You are missing the point.

The point is that in the two examples provided, setting an origin in G55 using G92 X0 Y0 **does not** affect G53 machine coordinates (correct behavior) whereas setting an origin in G54 using G92 X0 Y0 **does** affect G53 machine coordinates (incorrect behavior).

As a result, it is impossible to use work position coordinates (G54) in CNC.js:

[image]

shitcreek, 2019-08-13:
I stand corrected. G53 as a modifier should move relative to native space and when called by itself should switch back to native space.
This is what I've put together to get it to work. I've also made G28 set current position to the selected workspace.
[G53 - G59 - G28.zip](https://github.com/MarlinFirmware/Marlin/files/3495427/G53.-.G59.-.G28.zip)
If you wouldn't mind trying it out and let me know if functions as it should for you.

telleropnul, 2019-08-13:
Thank you for your time and effort - most appreciated!

I will have to reflash my CNC machine using the modified source code files provided.
I will report back my findings shortly...

telleropnul, 2019-08-20:
Sketch\src\gcode\calibrate\G28.cpp:210:22: error: 'homing_needed' was not declared in this scope

   if (!homing_needed() && parser.boolval('O')) {

exit status 1

Error compiling for board Arduino/Genuino Mega or Mega 2560.

shitcreek, 2019-08-20:
Not sure how you are getting that error.
Did you replace both cpp files with the ones I provided?

telleropnul, 2019-08-21:
I placed the two files in:
...\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\Marlin\src\gcode\calibrate\G28.cpp
...\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\Marlin\src\gcode\geometry\G53-G59.cpp
I then opened Marlin.ino sketch and started compile & upload.

I will run a diff on the files to inspect the changes and double check my steps.
More news soon + thank you for your help.

shitcreek, 2019-08-22:
Excellent. Some more testing of this code would be good. Like does it actually behave as it should when cnc'ing ie using the selected workspace offsets? Another thing to test for is does it respect the machine limits ie X Y and Z max?

I'll do up a pull request, and look into adding G10.

A favor to ask: since you seem to know your CNC stuff, would you mind looking over Marlin's documentation on the related functions such as G53. When I first authored that page I essentially just copied the information I was able to gather. It would be very helpful for someone knowledgeable on the subject matter like yourself to make corrections and clarifications. Please submit any suggestion to the documentation over at https://github.com/MarlinFirmware/MarlinDocumentation/issues. Much appreciated.

telleropnul, 2019-08-24:
I have successfully cnc'ed a number of test patterns that have the origin set to the center of the work piece rather than the bottom left corner.  As my machine uses double endstops, I cannot go into absolute negative X or absolute negative Y.   When using endstops, you can only move to positive X and positive Y machine coordinates.

When entering G54 + G92 X0 Y0 previously the machine would not go into absolute negative X and absolute negative Y as a result of the bug (G54 + G92 X0 Y0 = G53 would reset to X0 Y0 accidentally).  After your bugfix, G53 is no longer set to zero (hoorah!) and the original machine coordinates are preserved.  This allows me to go into absolute negative X and Y in G54 (which is perfectly fine and exaclty as it should be).  The double endstops correctly only reference G53 machine coordinates at all times and are not affected by G54-G59.

I agree peer review may be helpful before committing to source.  Your bugfix appears non-destructive in that it does not appear to break anything that might inpact others.

Documentation pages for G53 and G54-G59 appear to be correct.  There is no need to elaborate as that might only confuse and require constant updating.  It correctly states what it does.

GnawingPrinter, 2019-09-19:
I'm not entirely sure, but shortly reading over the bugfix i dont find any updating of the software endstops. Is this harmful?

telleropnul, 2019-09-19:
Endstops are not affected and keep functioning properly.  Endstops use machine coordinate system (G53) at all times.  This bugfix resolves an issue with work offset coordinate system G54.  When in G54 and entering G92 X0 Y0, this  would inadvertently set a new origin in both G53 and G54, which is not correct.  With the bugfix applied, G54 + G92 X0 Y0 now only sets a new origin for G54, which is the correct and desirable behaviour, similar to how G55 - G59 function.

driedell, 2020-02-01:
@telleropnul How did you get your axes widget to display 1, 0.1, 0.001... etc? Mine looks like this and the dropdown to select the distance drives me crazy:

[image]

telleropnul, 2020-02-03:
Dunno.  Possibly older version of CNC.js or perhaps a metric vs imperial thing?  I did not touch anything to make it behave differently.