MarlinFirmware/Marlin #14744: Compliation error LPC1768

davidcgu, 2019-07-27:
With latest version downloaded just today I have following compilation error:

`In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:30,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:205:39: error: missing binary operator before token "2016"
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)
                                       ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:134:35: note: in expansion of macro 'BOARD_BIGTREE_SKR_V1_3'
   #define MOTHERBOARD MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
                                   ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:309:46: note: in expansion of macro 'MOTHERBOARD'
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                              ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:38:5: note: in expansion of macro 'MB'
 #if MB(RAMPS_13_EFB) || MB(RAMPS_14_EFB) || MB(RAMPS_PLUS_EFB) || MB(RAMPS_14_RE_ARM_EFB) || MB(RAMPS_SMART_EFB) || MB(RAMPS_DUO_EFB) || MB(RAMPS4DUE_EFB)
     ^~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:205:39: error: missing binary operator before token "2016"
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)
                                       ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:134:35: note: in expansion of macro 'BOARD_BIGTREE_SKR_V1_3'
   #define MOTHERBOARD MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
                                   ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:309:46: note: in expansion of macro 'MOTHERBOARD'
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                              ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:40:7: note: in expansion of macro 'MB'
 #elif MB(RAMPS_13_EEB) || MB(RAMPS_14_EEB) || MB(RAMPS_PLUS_EEB) || MB(RAMPS_14_RE_ARM_EEB) || MB(RAMPS_SMART_EEB) || MB(RAMPS_DUO_EEB) || MB(RAMPS4DUE_EEB)
       ^~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:205:39: error: missing binary operator before token "2016"
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)
                                       ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:134:35: note: in expansion of macro 'BOARD_BIGTREE_SKR_V1_3'
   #define MOTHERBOARD MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
                                   ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:309:46: note: in expansion of macro 'MOTHERBOARD'
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                              ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:42:7: note: in expansion of macro 'MB'
 #elif MB(RAMPS_13_EFF) || MB(RAMPS_14_EFF) || MB(RAMPS_PLUS_EFF) || MB(RAMPS_14_RE_ARM_EFF) || MB(RAMPS_SMART_EFF) || MB(RAMPS_DUO_EFF) || MB(RAMPS4DUE_EFF)
       ^~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:205:39: error: missing binary operator before token "2016"
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)
                                       ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:134:35: note: in expansion of macro 'BOARD_BIGTREE_SKR_V1_3'
   #define MOTHERBOARD MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
                                   ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:309:46: note: in expansion of macro 'MOTHERBOARD'
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                              ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:44:7: note: in expansion of macro 'MB'
 #elif MB(RAMPS_13_EEF) || MB(RAMPS_14_EEF) || MB(RAMPS_PLUS_EEF) || MB(RAMPS_14_RE_ARM_EEF) || MB(RAMPS_SMART_EEF) || MB(RAMPS_DUO_EEF) || MB(RAMPS4DUE_EEF)
       ^~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:205:39: error: missing binary operator before token "2016"
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)
                                       ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:134:35: note: in expansion of macro 'BOARD_BIGTREE_SKR_V1_3'
   #define MOTHERBOARD MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
                                   ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:309:46: note: in expansion of macro 'MOTHERBOARD'
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                              ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:46:7: note: in expansion of macro 'MB'
 #elif MB(RAMPS_13_SF)  || MB(RAMPS_14_SF)  || MB(RAMPS_PLUS_SF)  || MB(RAMPS_14_RE_ARM_SF)  || MB(RAMPS_SMART_SF)  || MB(RAMPS_DUO_SF)  || MB(RAMPS4DUE_SF)
       ^~
compilation terminated due to -fmax-errors=5.
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfigPre.h:30,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp:26:
[... 171 lines cut ...]
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                              ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:44:7: note: in expansion of macro 'MB'
 #elif MB(RAMPS_13_EEF) || MB(RAMPS_14_EEF) || MB(RAMPS_PLUS_EEF) || MB(RAMPS_14_RE_ARM_EEF) || MB(RAMPS_SMART_EEF) || MB(RAMPS_DUO_EEF) || MB(RAMPS4DUE_EEF)
       ^~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:205:39: error: missing binary operator before token "2016"
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)
                                       ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:134:35: note: in expansion of macro 'BOARD_BIGTREE_SKR_V1_3'
   #define MOTHERBOARD MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
                                   ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:309:46: note: in expansion of macro 'MOTHERBOARD'
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                              ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:46:7: note: in expansion of macro 'MB'
 #elif MB(RAMPS_13_SF)  || MB(RAMPS_14_SF)  || MB(RAMPS_PLUS_SF)  || MB(RAMPS_14_RE_ARM_SF)  || MB(RAMPS_SMART_SF)  || MB(RAMPS_DUO_SF)  || MB(RAMPS4DUE_SF)
       ^~
compilation terminated due to -fmax-errors=5.
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.cpp.o] Error 1
======================================================= [ERROR] ``

Any idea?

davidcgu, 2019-07-27:
Note: with a previous version from 14 Jun I'm able to compile well

p3p, 2019-07-27:
You didn't follow the issue template for reporting bugs at all, we don't put that there just to be deleted you know. You also didn't supply your configs as requested in said template.

I'l answer grudgingly though as it's so simple.

```#define MOTHERBOARD MOTHERBOARD BOARD_BIGTREE_SKR_V1_3```
Check your configs.

closed 2019-07-27 by p3p

davidcgu, 2019-07-27:
Sorry just saw that, wrote mtherboard twice on config..... hahaha thanks a lot!