MarlinFirmware/Marlin #14747: [FR] Endstop interrupt for only Z axis

mikeshub, 2019-07-27:
### Description

I wanted the Z axis endstop on my SKR1.3 Creator Pro to be on an interrupt. There was only one pin suitable for endstop interrupt. To get around this I just commented out the code in setup_endstop_interrupts for the relevant X and Y defines. This worked great. I now get the improved precision of the Z axis endstop on an interrupt. The X and Y don't benefit from being on an interrupt nearly as much as the Z does. The SKR1.3 can't be the only board that for some reason didn't use interrupt pins for the endstops. The only interrupt capable pin I could find was at the servo connector. Buy commented out code isn't exactly "production ready."

### Steps to Reproduce

We need something better than this:

`
  #if HAS_X_MAX
    // #if !LPC1768_PIN_INTERRUPT_M(X_MAX_PIN)
    //   #error "X_MAX_PIN is not an INTERRUPT capable pin."
    // #endif
    // attachInterrupt(digitalPinToInterrupt(X_MAX_PIN), endstop_ISR, CHANGE);
  #endif
  #if HAS_X_MIN
    #if !LPC1768_PIN_INTERRUPT_M(X_MIN_PIN)
      #error "X_MIN_PIN is not an INTERRUPT capable pin."
    #endif
    attachInterrupt(digitalPinToInterrupt(X_MIN_PIN), endstop_ISR, CHANGE);
  #endif
  #if HAS_Y_MAX
    // #if !LPC1768_PIN_INTERRUPT_M(Y_MAX_PIN)
    //   #error "Y_MAX_PIN is not an INTERRUPT capable pin."
    // #endif
    // attachInterrupt(digitalPinToInterrupt(Y_MAX_PIN), endstop_ISR, CHANGE);
  #endif
`

mikeshub, 2019-07-27:
Actually this isn't working how I thought.

closed 2019-07-27 by mikeshub

Virtualight, 2019-08-21:
The servo pins on SKR1.3 are designed to be used for BLTouch z-probe which makes sense for it to be interrupt capable...