MarlinFirmware/Marlin #14752: [BUG] [bugfix-2.0.x] Weird stutter (knocking) during some travel moves on a cartesian machine

orcinus, 2019-07-28:
### Description

During some travel moves, and *only* travel moves, i get this weird occasional stutter that sounds almost like the nozzle dragging across a grid. So far, it only seems to happen with travel moves that include both X and Y axes, at a very sharp angle (close to one of the axes). Sometimes i’ll get a travel move in one direction that’s completely smooth, then the stutter in the opposite direction.

It’s not nozzle dragging, i’ve increased the hop to a comically high value to check and verified by looking at it while it was happening. It’s not mechanical, as i can’t reproduce it by moving the axes manually, or with manual G-code. I doubt it’s the slicer, as i’ve been using the same slicer and same version (KISSlicer 2 0.1.11 alpha from Feb 2018) for a very long time now (since i had very old 1.x Marlin builds on this machine).

It’s a simple cartesian machine (BigBox) with very conservative acceleration and jerk, on a RUMBA, TMC2100s.

Here’s a video (you’ll need to bring up the volume to hear it) where you can hear it NOT happening in one direction, then happening immediately after in the other direction:

https://www.youtube.com/watch?v=SJttZs4xOZ4

Any ideas on what might be causing this are welcome.
Apologies if this was already reported, i couldn’t find anything quite the same.

orcinus, 2019-07-28:
I’d gladly provide the relevant G-code, but since it usually happens somewhere in the middle of a layer, i’ve got no good way to find the G-code by eye. It also doesn’t happen on every layer, i’m guessing because the angles of the travel moves change because of the infill direction.

boelle, 2019-07-28:
do the TMC2100 have active cooling? also remember they run best at 24V 
if you dont have 24V you can use a step-up converter and feed the 12V directly to the VM pin

Scratchbuild52, 2019-07-29:
That knocking sound happened on my hypercube also and I discovered that I overtightened my belts and it stretched my belt to such an extent that the teeth on the pulleys does not fit the belt anymore. Check your belts for places where it may have stretched to a point where the pitch of the teeth is not what it should be. I could see it easily on my belts.

orcinus, 2019-07-29:
> do the TMC2100 have active cooling? also remember they run best at 24V
> if you dont have 24V you can use a step-up converter and feed the 12V directly to the VM pin

Yup, they do have active cooling, and they are running at 24V.

> That knocking sound happened on my hypercube also and I discovered that I overtightened my belts and it stretched my belt to such an extent that the teeth on the pulleys does not fit the belt anymore.

I've tried moving manually (as in, by hand) and via manual G1's and haven't managed to reproduce it, so i'm not 100% sure it's mechanical, but good point. It might only occur at a specific speed for some reason.

I'm using IGUS bearings on one of the axes, so for a while i was pursuing the possibility of it being the typical IGUS chatter. However, again, moving either axis by hand or via G1 does not reproduce anything similar, no matter how many times i've tried and no matter the speed.

boelle, 2019-07-29:
oh yeah,,, now i rememeber my build too (i have not finfished it)

i had to replace the 2 bearings in each end of the X axis (not on the carriage itself) with long versions as 2 short version had a tendency to grind and make funny noises

orcinus, 2019-08-04:
Did some more debugging, sadly, no change.
I’ve removed two of the bearings (so only one per axis now). I’ve tried reproducing the stutter again, both by moving the axes by hand, and via g-code. Nothing. No stutter of any kind.

I’ve examined the belts, to check if maybe it’s a belt rubbing against the edge of a pulley - again, nothing, everything’s fine.

I’m out of ideas, at least as far as hardware is concerned.

boelle, 2019-08-06:
I’ve removed two of the bearings (so only one per axis now). I’ve tried reproducing the stutter again, both by moving the axes by hand, and via g-code. Nothing. **No stutter of any kind.**

for me that seems like a fix, you say you have no stutter

orcinus, 2019-08-06:
I think you’ve misunderstood me…

> I've tried moving **manually** (as in, **by hand**) and via **manual G1's** and haven't managed to reproduce it, so i'm not 100% sure it's mechanical

>  I’ve tried reproducing the stutter again, both by moving the axes **by hand**, and **via g-code**. Nothing.

I can’t reproduce it *manually*. The original issue was never occurring when doing manual jogs and moves - only while printing. That issue still stands.

orcinus, 2019-08-06:
In the meantime… I’ve tried a different slicer (Cura vs. KISSlicer) to verify if the slicer is to blame - same result for both. So that’s not it either.

I’m now down to swapping out the drivers, in a (faint) hope that maybe one of the TMC2100s is faulty in some bizarre way.

pseudex, 2019-08-08:
On another issue I read knocking can happen if the controller is doing some spi communication during the move. While communicating the controller can not change the step pin.
You could try to disable some of the reporting debug functions of the driver.

orcinus, 2019-08-08:
> On another issue I read knocking can happen if the controller is doing some spi communication during the move. While communicating the controller can not change the step pin.

Hmmm. I'm not printing from the SD card, and i don't have any other SPI peripherals configured or connected. Maybe it's trying to poll it regardless? Will look into it, thanks!

In other news, tried swapping drivers, same thing, no change.

gloomyandy, 2019-08-08:
@orcinus the SPI comment is with regard to the way that Marlin will poll each of the drivers every few seconds to check for thermal and other errors. It does this if you have MONITOR_DRIVER_STATUS defined. I'm not aware of this causing any issues (and would be surprised if it does), but who knows...

shitcreek, 2019-09-17:
Can you trace where the sound is emanating from? motor, bed area or on the board?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle