MarlinFirmware/Marlin #14753: BL Touch Mesh Inaccurate

Neiljclements, 2019-07-28:
I’d been having problems with my probes, both inductive and BL Touch on Marlin 1.1.x (final). The mesh would be useless - too close in places, too far away in others.
I upgraded to Marlin 2.0 (release 19.07.19 onwards) and the inductive sensors created a perfect mesh and the first layers were perfect, with no other changes. Unfortunately the BL Touch still had the same issue. 
I have since swapped out the BL Touch sensors for NPN NO inductive because they work really well under 2.0.
I have replicated the problem by going back to 1.1.x (final) and the problem is there again.
I appreciate no further development is being done on 1.1.9, however as the issue continues with the BL Touch under 2.0, I figured it was worth mentioning.
I have discussed this in various forums and I’m not the only person to discover this issue.

thisiskeithb, 2019-07-28:
Help us help you.

Please fill out the [issue template](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/issue_template.md) as requested when you created the issue and attach your configs.

boelle, 2019-07-28:
what @thisiskeithb said ^^^

Neiljclements, 2019-07-28:
Description
Steps to Reproduce

    1. Install Marlin 1.1.x and probe bed.  Run single layer test print and first layer is terrible - too close in some places, too far in other places.
    2. Install Marlin 2.0.x and probe bed.  Run single layer test print and first layer is perfect.

Expected behavior: Perfect mesh with both 1.1.x & 2.0.x

Actual behavior: 1.1.x doesn't seem to create an accurate mesh.

Further notes: The issue was only solved when using an inductive sensor.  When using a BL Touch, neither version would create a mesh good enough to put down a nice first layer.

[2.0.x_ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3439184/2.0.x_ConfigFiles.zip)
[1.1.x_ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3439185/1.1.x_ConfigFiles.zip)

thisiskeithb, 2019-07-29:
Your `bugfix-2.0.x` configs are a bit dated. Can you [download the latest `bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) and see if the issue persists?

Are you using the stock Crealty board in your CR-10S? If so, you should change the `MOTHERBOARD` define to `BOARD_RAMPS_CREALITY` to match the [CR-10S configs](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/CR-10S) (or just start over with those).

Which version of the BLTouch do you have? It would also help if you broke out the configs by type (`BLTOUCH` &  `FIX_MOUNTED_PROBE`) to make sure you're setting configs up properly.

boelle, 2019-08-06:
@Neiljclements tried what @thisiskeithb suggested?

shitcreek, 2019-09-17:
Cleaning house. LMK if problem still persists and issue needs to be reopened.

closed 2019-09-17 by shitcreek