MarlinFirmware/Marlin #14756: [BUG]  STM32F407ZGT6  Neopixel Support unable to compile

ledfreak3d, 2019-07-28:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RudyBenner, 2019-07-28:
[image]

ledfreak3d, 2019-07-28:
Yep i know because  define neopixel is commented in.my version uncomment and see what happens

RudyBenner, 2019-07-28:
This is what you sent.

[image]

RudyBenner, 2019-07-28:
Yup

[image]

ledfreak3d, 2019-07-28:
[image]

mikeshub, 2019-07-30:
You need to edit your platformio.ini. In STM32F4 neo pixel is in the ignore. This is because the library does not really support 32 bit targets. If you want to help add support for neopixels to the STM32F4 here is what needs to be tested. The Adafruit NeoPixel library was just updated to work with LPC1768 targets. I'm pretty sure this is an easy fix, but I don't have an STM32F4 board to test this with. Here is the relevant code for anyone who feels like trying the fix:

https://github.com/adafruit/Adafruit_NeoPixel/pull/197/commits/0f78242a630d6d0ea27be42c9f830b9d17e9bc00

What needs to happen is where it has #elif defined(TARGET_LPC1768) that needs to be changed to an or with the correct target for the STM32F4. 

It is just two steps to do this.

* Edit platformIO.ini to have STM32F4 use the code from Adafruit's github. This should make it download the library automatically..
* Navigate to Adafruit_NeoPixel.cpp and add the STM32F4 target into the correct define

ledfreak3d, 2019-07-30:
> adafruit/Adafruit_NeoPixel@0f78242

Ill give it a go and update you

ledfreak3d, 2019-07-30:
Ok  its looking like this is going to work was abl to compile atleast let me check if they also work

ledfreak3d, 2019-07-30:
Hey Mike 

 

It does compile without errors now

But no light yet tried several of the  spare i/os but nothing yet

ledfreak3d, 2019-07-30:
Changed all of them compiles ok but no output on any pins

ledfreak3d, 2019-07-30:
Hey Mike

 

 

Ok so i twill compile but no output at all 

See modified files for more info

mikeshub, 2019-07-30:
I'm not seeing the files.

ledfreak3d, 2019-07-31:
Here you go fors ome reason they where not included but they where in my send email very wierd

ledfreak3d, 2019-08-04:
not been able to get it working yet

Evg33, 2019-09-13:
It's worked ! :)

https://youtu.be/uGW9yKVrxyU

Evg33, 2019-09-13:
https://github.com/adafruit/Adafruit_NeoPixel/pull/206#issuecomment-531297764

Evg33, 2019-09-13:
ladyada merged commit ccba64f into adafruit:master 1 hour ago

Evg33, 2019-09-13:
https://github.com/MarlinFirmware/Marlin/issues/14484#issuecomment-531298624

boelle, 2019-09-24:
@ledfreak3d still having issues?

ledfreak3d, 2019-09-24:
Hey Bo Nope working great now

boelle, 2019-09-24:
oki closing then

closed 2019-09-24 by boelle

ledfreak3d, 2019-09-24:
Thsnks Bo