MarlinFirmware/Marlin #14762: [BUG] U8glib compile error unknown type name 'class'

mattimm, 2019-07-28:
### Description

When I add the line  "#include <U8glib.h>" to configuration.h, I get errors in U8glibPrint.h and U8glibPrintable.h saying "unknown type name 'class'".

### Steps to Reproduce

I typed #include <U8glib.h> and I let VSCode autofill the line which points to the library in LPC1768. When I click on the build option in Platformio, it runs for around 4 seconds before giving me the error message 

".piolibdeps\LPC1768\U8glib-HAL_ID1932\src/U8glibPrintable.h:26:1: error: unknown type name 'class'
 class U8glibPrint;
 ^~~~~
.piolibdeps\LPC1768\U8glib-HAL_ID1932\src/U8glibPrintable.h:34:1: error: unknown type name 'class'
 class U8glibPrintable
 ^~~~~"

I read online that this might be caused by circular "#include" but I'm not good enough at coding to find where the error is lol. Has anyone else experienced this? Judging from the fact that I can't really find anyone else having longstanding errors like this, I'm assuming its a dumb typo I've made somewhere.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3439646/configs.zip)

thisiskeithb, 2019-07-29:
What's the purpose of including the U8glib manually? It should be covered in the library dependencies for your board through the `LPC1768` environment in [`platformio.ini`](https://github.com/MarlinFirmware/Marlin/blob/eb7840dabb5852a3bc7bfe1a2466ad691d2bdae4/platformio.ini#L167).

mattimm, 2019-07-29:
> What's the purpose of including the U8glib manually? It should be covered in the library dependencies for your board through the `LPC1768` environment in [`platformio.ini`](https://github.com/MarlinFirmware/Marlin/blob/eb7840dabb5852a3bc7bfe1a2466ad691d2bdae4/platformio.ini#L167).

I'm including it because the display only shows a blue screen without it, and I saw some people on the RepRap cite fixing that by manually writing the include.  I'm not sure if that is the screen solution, though.

closed 2019-07-30 by mattimm

thisiskeithb, 2019-07-30:
> *edit: I got the screen to work, making this issue irrelevant. I just needed to modify the ribbon cables connecting the screen, so it wasn't a driver issue

Glad you got it working!