MarlinFirmware/Marlin #14764: [BUG] (The LCD12864 screen shows garbled characters)

makerbase-mks, 2019-07-29:
### Description

LCD12864 runs fine on Marlin 1.1.3, 1.1.5, 1.1.8, 2.0, but there will be garbled screens in 1.1.9. After adjusting the delay time, the test can be used normally. We to provide our revised content in here, I hope the Marlin team can test it, if the situation is the same as we described, we hope that you can fix it in bug-fix-1.1.x, thank you!

### Steps to Reproduce

1. LCD12864 runs on Marlin-1.1.9 or Marlin-bugfix-1.1.x.
2. The LCD12864 screen shows garbled characters.
3. Adjusting the delay time in Marlin-1.1.9 or Marlin-bugfix-1.1.x.
[image]
4. Test can be used normally.

**Expected behavior:** the LCD12864 can be used normally

**Actual behavior:** The LCD12864 screen shows garbled characters

#### Additional Information

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3440663/configuration.zip)

boelle, 2019-08-06:
bug-fix-1.1.x is not updated anymore, only bug-fix-2.0.x

shitcreek, 2019-08-07:
In 2.0, the parameters are still the same:
`#elif F_CPU == 16000000
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(63)
#else`

Perhaps there is something else at play? In anycase, since it works in 2.0 and 1.x is no longer being worked on, I am closing this issue.

closed 2019-08-07 by shitcreek