MarlinFirmware/Marlin #14765: adding some code on marlin

henryprobosantoso, 2019-07-29:
i want to add some code like "analogWrite(PIN, Value)" whenever the extruder move, but i am confused about where should i place the code, please tell me how to do it.
thanks.

reloxx13, 2019-07-30:
tell us what u want to do.
there are gcodes to trigger a pin.

henryprobosantoso, 2019-07-30:
i just want to replace Extruder stepper with motor DC, and to control it i use PWM, so i just want to place code that generate PWM whenever the extruder move. But i dont know where to place it
thanks ind advance

henryprobosantoso, 2019-09-03:
yeay thanks @shitcreek , btw finally i can add some PWM function, with M106, thanks all, what i did just give a simple analogWrite function inside gcode_M106 function on marlin_main.cpp file, thanks all

closed 2019-09-03 by shitcreek