MarlinFirmware/Marlin #14766: Possible Bug autolevel

Louis-Cypher, 2019-07-29:
Starting mid of May version of Marlin 2.0.x I cannot autolevel anymore. I tried to change various parameters starting from Z- probing speed to z-axis Minimum positions etc but sadly cannot get autobed leveling working again.

Others have reported the same behavior: https://reprap.org/forum/read.php?415,852720,853793#msg-853793

## Machine
Core XY running most recent datecode of Marlin 2.0.x. Same behaviour if running most recent 1.x Version. Used bilinear probing as well as linear. Inductive probe.

## What happens
G28 works fine
G29 starts leveling but more or less arbitrarily  stops with message "probing failed". I can see that the z-probe is triggered correctly. I enabled debugging but I cannot find anything of help in the debugging output. Please find debugging output as well as configuration attached. 

[leveling_debug.txt](https://github.com/MarlinFirmware/Marlin/files/3440955/leveling_debug.txt)

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3440971/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3440972/Configuration_adv.h.txt)

robbycandra, 2019-08-01:
It is not a bug.

You should raise your bed. 

It is fail because the probe cannot reach your bed. 
The correction should not bigger than 2 mm.

Another workaround is to change 
```
Z_PROBE_LOW_POINT -2
```
to bigger negatif number like -3

``
>>> do_blocking_move_to(70.00, 39.00, -2.00)
<<< do_blocking_move_to
echo:busy: processing
  current_position=(70.00, 39.00, -2.00) : sync_plan_position
  current_position=(70.00, 39.00, -2.00) : <<< do_probe_move
SLOW Probe fail!
  current_position=(70.00, 39.00, -2.00) : <<< run_z_probe
```

closed 2019-08-07 by shitcreek

Louis-Cypher, 2019-08-12:
Thanks a lot. That helped.

Louis-Cypher, 2019-08-13:
Sorry, but I am not convinced that this behavior is correct. To which respect should I raise the bed? I am using the same probe for homing and bed leveling. Hence, there should be no difference in Z-height. I changed the value to -3 and have the same problem again.

Louis-Cypher, 2019-08-13:
For documentation purpose: During trying to fix the problem I misinterpreted and changed Z_PROBE_LOW_POINT to -3. This causes an offset of 3mm to the probe causing the problem detailed above (BTW: to me the comment behind the variable was not clearly understandable; I discovered the solution by searching something probe related in the config)