MarlinFirmware/Marlin #14767: [BUG] BIGTREETECH SKR PRO V1.1 - only boots when serial host connected

NickMilton, 2019-07-29:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Ludy87, 2019-07-29:
ok

golfromeo-fr, 2019-07-29:
can it be related to USB configuration of SERIAL_PORT 1/2 ? or the pin layout is not exactly the same in the 2 repositories?
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/18

I will test my MKS LCD 12864 which doesn't light up but I haven't checked if it is better when I open pronterface. Until I receive my BIGTREETECH LCD 12864

NickMilton, 2019-07-30:
> bigtreetech/BIGTREETECH-SKR-PRO-V1.1#18

Ahh

Good pick up. I set up the port configs to match the Bigtreetech provided
build - assuming they were correct. I'll try the other way round. Will not
be till weekend  - travelling.

dch1921, 2019-07-30:
I believe the first serial port needs to be -1. The second can be 1 or 6 if you are using an esp01 wifi

golfromeo-fr, 2019-07-31:
> I believe the first serial port needs to be -1. The second can be 1 or 6 if you are using an esp01 wifi
is it related to
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/9
?

dch1921, 2019-07-31:
> > I believe the first serial port needs to be -1. The second can be 1 or 6 if you are using an esp01 wifi
> 
> is it related to
> [bigtreetech/BIGTREETECH-SKR-PRO-V1.1#9](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/9)
> ?

Not directly related but from what I understand Marlin uses serial port 0 or -1. When this is defined as the second serial port, Marlin doesn't fully load until a USB connection is established.

NickMilton, 2019-08-02:
Problem solved - thanks for your suggestions golfromeo-fr and dch1921.

All I did was set....
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

and it all came good. Not a bug in Marlin - was a limitation in user knowledge. The Bigtreetech board is a good one too. No complaints there at all.

Cheers

closed 2019-08-02 by NickMilton

farmstar72, 2019-08-07:
I'm having the same problem but this fix didn't work for me - anything else I should look at?

Thanks,
Scott

NickMilton, 2019-08-07:
Hi farmstar72

1st thing I would do is compare your config files to mine.....and see what the differences are. Could be a hardware/wiring issue?

farmstar72, 2019-08-07:
Hi Nick, I used your config file and still have this issue. My board is exactly as shipped with no drivers installed. I did try removing the driver jumpers to no effect.

Thanks,
Scott

NickMilton, 2019-08-07:
Scott - I 'grew up' on Repetier f/w so still llearning how Marlin behaves (so far I like it very much). I am wondering if you need to connect up thermistors, LCD etc in order for Marlin to fully boot. Just a thought. Also I have not installed the WiFi module yet. Other than that mine is stock too..

farmstar72, 2019-08-07:
Nick, I've only been at this a few months. I started with Arduino IDE and an MKS Gen L board. Then I moved to VsCode to use Marlin 2.0 with an SKR V1.3 board. I've compiled Marlin 2.0 with VsCode/Platfomio for MKS Gen L, Ramps 1.4, Re-Arm and SKR V1.3. Each have had their challenges - ie; too long of file name, anti-virus issues, firewall, etc. - but I've made it through them all (using the same RepRapDiscount Full Graphic Smart Controller). This one has me kinda stumped though.

Thanks for your input,
Scott

farmstar72, 2019-08-07:
WooHoo! - I got it to work!
Here's what I learned 
You have to copy the firmware to the SD card and not upload using the USB. The USB lets you upload using a Com port but with inconsistent results. I finally gave up and started my trial and error but using the SD card plus your configuration and it works now.

Thanks,
Scott

NickMilton, 2019-08-07:
Good one. I have only ever been uploading using the microSD card - which explains our different experiences.

farmstar72, 2019-08-07:
I've uploaded with USB to all my other boards, including 32 bit ARM's. The 32 bit ARM boards have SD cards (Re-Arm/Ramps and SKR V1.3) that show up as drive D: when I connect my USB cable and the card is inserted in the board - With those boards, I can either paste firmware.bin onto the SD or flash the board over USB. When I first got the SKR Pro, I thought it was defective because I couldn't see the SD card like that. The seller set me straight there but I was still following the habit of flashing over USB since I could. Glad I got that sorted. 
To summarize to anyone that has this same issue;
1. Modify config.h as above with:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
2. You must copy and paste the firmware.bin file directly to the SD card then insert it into the SKR Pro and power up.
- Also, something interesting I noticed; Once the firmware is flashed to the board from the SD card, the firmware.bin file is deleted from the SD cardand only firmware.cur remains.