MarlinFirmware/Marlin #14769: [BUG] X-axis prints concave.

Scratchbuild52, 2019-07-29:
Hi there,

I

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-07-29:
I recommend using bugfix 2.0 because if this is a bug, it might have already been addressed.
Have you printing the model rotated by 90* and see if you're getting the same result? You could try swapping x motor for y (more appropriately delta A motor and delta B motor) instead of replacing them by switching their plugs. You could also try swapping the drivers.

Scratchbuild52, 2019-07-30:
Will do, just picked up another problem, have to get a part printed before I can try again.

RudyBenner, 2019-07-30:
Does the bed have to be square? 
Line 1050-1051

thinkyhead, 2019-07-31:
If the X-axis smooth rods are at all twisted in relation to each other it can cause the gantry to twist, which moves the nozzle slightly in the Y direction. Look very carefully from the side, or set up a fixed camera, and observe whether the nozzle shifts at all as the carriage moves along the X axis.

thinkyhead, 2019-07-31:
Rather than attaching the STL file, you should attach your `.gcode` file.

thinkyhead, 2019-07-31:
Just in case this is not a bug, be sure to take advantage of the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

Scratchbuild52, 2019-08-01:
That would have been the next thing I checked, but I am busy with projects
at work and cannot look at it right now.
Will let you know as soon as I can and give feedback.

lpgreini, 2019-08-01:
Hi,
I had the very same issue with my Hypercube Evolution. I tested firmwares 1.1.8, 1.1.9, 2.0.x bugfix. The issue was there on any firmware I tested. So I tested Repetier Firmware and, well, the issue was there too ;-)
So I began to switch hardware and the faulty part was the belts. I wanted to have the best and used those steel core belts. Those were just too hard for this kind of setup. I switched back to nylon core rubber belts and everything is fine now.
Give it a try ;-)

lpgreini, 2019-08-01:
I forgot to say: no matter what belts you are using, it is worth switching them to others. Even if it's not steel core or something other special. They just may be of bad quality...

Scratchbuild52, 2019-08-02:
Hi Lpgreini,
                     Thanks for your reply. At least I feel I have
somewhere to start. I actually do have the steel core belts on my printer,
well not now, I have pulled almost my whole printer apart to check it for
sqaureness and dimensional problems. I have quite a bit of building to do
to get back at the printing stage :-) , but I will definitely try the
switching of the belts.
Thanks again.

closed 2019-08-05 by thinkyhead

Scratchbuild52, 2019-08-07:
Just feedback: It was the belts that was the problem