MarlinFirmware/Marlin #14770: [BUG] Can't compile for LPC1769

klcjr89, 2019-07-29:
When trying to build for my azteeg x5 mini, I now get compile errors whereas it used to work perfectly fine before.

```
Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1769; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.80KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: framework-arduino-lpc176x 0.1.3, toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
LibraryManager: Installing U8glib-HAL
Using cache: /Users/Ken/.platformio/.cache/cf/a54ce8ed1b234ada291ce966617e3dcf
Unpacking...
U8glib-HAL @ 0.4 has been successfully installed!
Looking for TMCStepper library in registry
Found: https://platformio.org/lib/show/5513/TMCStepper
LibraryManager: Installing id=5513 @ <1.0.0
Using cache: /Users/Ken/.platformio/.cache/c0/6785fb5c4dd051fa5e3db66eda6c10c0
Unpacking...
TMCStepper @ 0.4.5 has been successfully installed!
LibraryManager: Installing Adafruit NeoPixel
Using cache: /Users/Ken/.platformio/.cache/15/b58c51ea6e656569f518a720d934cc15
Unpacking...
Adafruit NeoPixel @ 1.1.8 has been successfully installed!
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.4.5
|-- <Adafruit NeoPixel> 1.1.8

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pioenvs/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/DebugMonitor_LPC1768.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL_spi.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL_timers.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/MarlinSerial.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/fast_pwm.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/include/digipot_mcp4451_I2C_routines.c.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/main.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_flash.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_I2C_routines.c.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.c.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_hw_spi.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_hw_spi.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/DebugMonitor_LPC1768.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/DebugMonitor_LPC1768.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL_spi.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL_spi.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_flash.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_flash.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/main.cpp.o: No such file or directory
*** [.pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/main.cpp.o] Error 1
OSError: [Errno 2] No such file or directory: '/Users/Ken/Marlin/.pioenvs/.sconsign.dblite':
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1346:
    _exec_main(parser, values)
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1309:
    _main(parser)
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1091:
    nodes = _build_targets(fs, options, targets, target_top)
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1283:
    jobs.run(postfunc = jobs_postfunc)
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/Job.py", line 113:
    postfunc()
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1280:
    SCons.SConsign.write()
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/SConsign.py", line 109:
    syncmethod()
  File "/Users/Ken/.platformio/packages/tool-scons/script/../engine/SCons/dblite.py", line 129:
    self._os_unlink(self._file_name)
Exception OSError: (2, 'No such file or directory', '/Users/Ken/Marlin/.pioenvs/.sconsign.dblite') in <bound method dblite.__del__ of <SCons.dblite.dblite object at 0x10f39ae10>> ignored
========================== [ERROR] Took 4.68 seconds ==========================
```

p3p, 2019-07-29:
This looks like a PlatformIO issue not Marlin, you can try deleting all the PlatformIO cache folders such as `/Users/Ken/.platformio` and `Marlin/.pioenvs` etc.

klcjr89, 2019-07-29:
@p3p I've already tried deleting those folders and cleaning and building again, still the same error.

p3p, 2019-07-29:
You should probably direct the issue to the platformio community, and deleting everything and trying again is about the best advise I can offer when it goes wrong, I've no idea why it can't create the files it wants to.

klcjr89, 2019-07-29:
I've found these sorts of errors to occur when the codebase or library dependencies are changed in Marlin.

klcjr89, 2019-07-29:
Just uninstalled and reinstalled Atom and the issue is still there. Please help.

closed 2019-07-29 by klcjr89

p3p, 2019-07-29:
I would move to VSCode if you can it's never been that stable, PlatformIO is moving away from Atom and they suggest VSCode by default on their website.

klcjr89, 2019-07-29:
@p3p I just switched to VSCode and only took about 5 minutes to get up and running. Thanks for the tip!