MarlinFirmware/Marlin #14773: [BUG] mks sa

zuki07, 2019-07-29:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

zuki07, 2019-07-29:
I have a mks sbase with a tft32 touchscreen.  The problem I am experiencing is when I hook up the board to my computer (windows 10 pro 64 bit) via usb cable I can open the sd without any problem however I do not have a com port displayed in device manager.  I have tried multiple changes in the firmware and have had no success.  I have also looked at many other reports on here and tried a few solutions that worked for someone else but I still haven't had any success.  Everything else works great including my leds.  I have also installed smoothieware firmware and was able to get a com port and run my 3d printer on pronterface.  Any help would be much appreciated.  I have also attached my config.h and config_avd.h.  I use atom pio to build and upload on my sd card.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3444442/Configuration.txt)

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3444444/Configuration_adv.txt)

RudyBenner, 2019-07-30:
I would try commenting out Line 116.
Seems to me you may need to load the CH340 driver.

Probably not related to the problem, especially since you are only using 1 extruder.
Line 670 719 726 734 in configuration.h

zuki07, 2019-07-30:
I have tried commenting out line 116 and still no com port appears.  The device manager is asking for a driver I have yet to figure out how to properly load one.  Where would I find the CH340 driver?

zuki07, 2019-07-30:
I found and downloaded the driver however when I try to install it says that it is already installed.  Poked around and yep, the driver is already installed on my computer but for some reason not reading the marlin usb device.

zuki07, 2019-07-31:
Update:  I hooked up my laptop to my 3d printer and I get a com port which allows me to connect to pronterface and all is good.  My laptop is running on windows 10 home as where my desktop computer is running windows 10 pro.  I would like to plug my 3d printer in to my desktop and have it connected to pronterface via usb.  My desktop does have the ch341 drivers installed as where my laptop does not.  This does not make sense so me.  Any tips would be much appreciated.

p3p, 2019-07-31:
Windows 10 does not require drivers for CDC Serial devices, (previous versions do), so it could possibly be a conflict.

It should "just work" on Windows 10 as it did on the laptop.

zuki07, 2019-08-01:
Right, I'm just now sure where the conflict could be.

p3p, 2019-08-01:
I would try completely removing the driver you have installed and see if that helps.

zuki07, 2019-08-01:
I was not able to install a driver for marlin usb device but am tryin to get completely rid of ch341 driver on my desktop to see if it helps.  So far windows will not let me, so time for some research and see if I can get rid of the ch341 driver seeing that my laptop does not have this driver installed.

zuki07, 2019-08-08:
After many hours of trying different things I updated my bios and an update in windows 10.  It finally works and all drivers are up to date and working.

closed 2019-08-08 by zuki07