MarlinFirmware/Marlin #14775: [BUG] Motors buzz when HYBRID_THRESHOLD is activated

juchong, 2019-07-30:
### Description

Motors begin to buzz (and not move) during a homing operation, specifically during the HYBRID_THRESHOLD transition time. My printer uses TMC2130 drivers and an SKR1.3 board. I'm running the latest Marlin 2.0.x development build and am trying to use native 256 step mode without interpolation. It's important to note that reducing the number of steps to 16 from 256 allows HYBRID_THRESHOLD to operate properly. If HYBRID_THRESHOLD is disabled and the printer is operated in SpreadCycle mode only, then the printer correctly operates. 

### Steps to Reproduce

1. See attached configuration. 
2. Assuming the hardware mentioned above is used, executing a `G28` command should force the printer to slowly (StealthChop) move the axes until the limit switches click.
3. Once the limit switches are tripped, a movement that should be executed in SpreadCycle mode will cause the motors to buzz loudly. 

**Expected behavior:** Fast movement and less buzzing

**Actual behavior:** Lots of buzzing and no movement

#### Additional Information
Configuration files attached. 
[buzzing_config.zip](https://github.com/MarlinFirmware/Marlin/files/3445472/buzzing_config.zip)

mikeshub, 2019-07-30:
Z axis always homes in spread cycle when hybrid threshold is enabled. This falls under the "it's a feature not a bug" category. You might make a feature request for this to be able to be turned on or off. I only print in stealth mode on my CR10s and Creator Pro. YMMV obviously, but you might try just stealth to start with if you haven't already.

R0GUEx, 2019-07-30:
This sounds for me as if the drivers are not receiving enough current. Have you tried adjusting the vref on your motors? Is your powersupply delivering enough power to the stepper drivers?

juchong, 2019-07-30:
@mikeshub I issued an M502 followed by an M500 before testing. I also power cycled the printer after loading new firmware images. Could I take a look at your config files? I suspect it's a tiny configuration change that's unique to my setup.
@R0GUEx The printer moves in and out of spread cycle mode as expected when using 16 microsteps. It also functions properly if I disable StealthChop, so I suspect that insufficient current isn't the issue.

teemuatlut, 2019-07-30:
@mikeshub The TMCs don't support any gcodes that alter the microstepping setting so M500 doesn't do anything. M502 re-initializes the drivers so maybe that's what helped you. Powering off-on is still a good practice to clear the drivers though.

@juchong Can you try without XY quick homing?

mikeshub, 2019-07-30:
@teemuatlut M500 does indeed do something which is to save the correct steps per mm. There isn't a gcode for microstepping, but M92 is pretty important when changing microstep modes if the user didn't change the defines to the correct steps per mm when switching microstepping modes. 

@juchong Here are the configs and pins headers.

[SKR.zip](https://github.com/MarlinFirmware/Marlin/files/3448628/SKR.zip)

boelle, 2019-09-24:
@juchong is the issue still there?

juchong, 2019-09-28:
@boelle I wasn't ever able to get native 256 microstepping working on my printer, even with the profile @mikeshub linked above. It looks like even his profile isn't using 256 microstepping, so he wouldn't have run into the issue either. I eventually reverted back to extrapolating 16 to 256 steps and moved on, but the bug is still present.

boelle, 2019-10-21:
@juchong did you use 24V or 12V ? just a question that came to my head

juchong, 2019-10-21:
@boelle I'm using 24V

boelle, 2019-10-21:
oki, TMC suggest that as the voltage, so one thing less on the list of possible things

is the buzz loud?

juchong, 2019-10-21:
Here's a video of the buzz @boelle: https://youtu.be/f0MR3OqHhQ8

boelle, 2019-10-21:
wow, that is a giant BUZZ @teemuatlut any idea what that could be?

teemuatlut, 2019-10-21:
Sounds like trying to go too fast.

boelle, 2019-10-21:
makes sense yes, either to fast or to fast and to little amps

juchong, 2019-10-21:
@teemuatlut Not very likely. I was running homing the machine. Keep in mind that reverting the settings to interpolating 256 steps from 16 works well.

boelle, 2019-10-21:
try home slower anyway, the more microsteps the less torque you have and hence you might need to do it slower

if anythng it can rule out that you are running to fast

boelle, 2019-10-30:
@juchong did homing slower help? or is the issue still the same?

boelle, 2019-11-03:
will close this one as it seems that @juchong dont have a problem anymore

we can reopen if multiple can confirm there is an issue with marlin

closed 2019-11-03 by boelle