MarlinFirmware/Marlin #14777: [BUG] STM32 #if (PC8 == PB0) condition passes no matter of I/Os

trouch, 2019-07-30:
@tpruvot 

Try this snippet somewhere in your sources, such as config file, with an STM32F1 build:
```
#if PC8 == PB0
   #error "Something weird !"
#endif 
```

You should have error thrown very quick. This issue avoids to use any pin for Ex_AUTO_FAN_PIN, CONTROLLER_FAN_PIN, and such since the pre-compiler fails when comparing them with FAN_PIN at SanityChecks.

thisiskeithb, 2019-07-30:
Yep: See https://github.com/MarlinFirmware/Marlin/issues/14711 and https://github.com/MarlinFirmware/Marlin/issues/14732.

thisiskeithb, 2019-07-30:
@tpruvot: There’s definitely something wonky going on with STM32F1.

tpruvot, 2019-07-30:
indeed, there are enums... not defined constants
https://stackoverflow.com/questions/34677148/why-does-the-c-preprocessor-consider-enum-values-as-equal

tpruvot, 2019-07-30:
i think i found a solution... but unsure it will work on all STM32F1 see #14785

trouch, 2019-07-31:
thanks @tpruvot that works for me now 👍 
closing that issue.

closed 2019-07-31 by trouch