MarlinFirmware/Marlin #14778: [BUG] #define PIDTEMP disturbs serial communication

kkaleta, 2019-07-30:
### Description
SF:
Marlin 2.0.x downloaded 30.07.2019
HW:
Motherboard SKR 1.3 from @bigtreetech

Exact the same problem as reported by @cinealfa #14741 

After about half minute, SKR stops communication via serial port. What interesting, SKR is ain't halted, it's just stop talking.

### Steps to Reproduce

1. Compile Marlin 2.0.x with uncommented `#define PIDTEMP` 
2. Upload to a SKR 1.3 motherboard (and restart)
3. With the Pronterface, run gcode `M303 E0 S200 C8`

**Expected behavior:** [What you expect to happen]

-temperature chart fluctuate
-PID autotune result on Pronterface command line window

**Actual behavior:** [What actually happens]

-after about 20 secs, temperature chart stops on 60* When I hit Disconnect and Connect again, temperature is way higher, it shows about 200* and after half minute, chart it`s freezing again.

#### Additional Information

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3448272/Marlin.zip)

shitcreek, 2019-07-30:
Have you tried this suggestion from that issue:
> When you enabled it did you remember to load the default settings or set the PID gains? That happened to me early today when I turned on bed PID because the gains loaded as zero. M502 then M500 should sort you out or use M304 then M500. Be warned the current state of the autotuner might result in an over tuned bed that has temp oscillations. YMMV. Check this PR for more info:
> 
> #14746
> 
> How did the autotune settings work for you in their current form?

thisiskeithb, 2019-07-30:
Your configs aren't the latest version (they are missing [PID changes](https://github.com/MarlinFirmware/Marlin/commit/1579091c206c14bbf7c88186fae07ccddc001a36) from July 28), so download the latest `bugfix-2.0.x` code/transfer your config settings over manually and see if the issue persists.

I just ran a PID autotune on the hotend & bed for my SKR 1.3'd Ender-3 using the latest `bugfix-2.0.x` commit (https://github.com/MarlinFirmware/Marlin/commit/c1b99c43b6e3aec37dab84cd34357337f2c845fb) via OctoPrint without errors/serial comms disconnecting, so there's probably something else going on.

kkaleta, 2019-07-30:
> Your configs aren't the latest version (they are missing PID changes from July 28 at least)
You sure about that? My Version.h shows   `#define STRING_DISTRIBUTION_DATE "2019-07-30"` As I said, I've downloaded package today.

closed 2019-07-30 by kkaleta

thisiskeithb, 2019-07-30:
> You sure about that?

Yes, that's why I mentioned copying your settings over manually since you're missing the `MIN_POWER` define in `Configuration.h` that was added in https://github.com/MarlinFirmware/Marlin/pull/14746.

kkaleta, 2019-08-01:
Now I understand this situation. My very first step with a fresh Marlin is a copying the files from config\examples\Creality\Ender 3 to root\Marlin. Ergo -this example configuration.h is incomplete.