MarlinFirmware/Marlin #14780: [BUG] BAUD_RATE_GCODE ERROR COMPILING

Erokawa, 2019-07-30:
### Description
I tried to compile having uncomment #define BAUD_RATE_GCODE with atom and give me a error

Marlin\src\gcode\config\M575.cpp: In static member function 'static void GcodeSuite::M575()':
Marlin\src\gcode\config\M575.cpp:63:29: error: 'class CDCSerial' has no member named 'end'; did you mean 'read'?
       if (set0) { MYSERIAL0.end(); MYSERIAL0.begin(baud); }
                             ^~~
                             read
*** [.pioenvs\LPC1768\src\src\gcode\config\M575.cpp.o] Error 1
### Steps to Reproduce

1. comment #define BAUD_RATE_GCODE
2. try to compile on atom

**Expected behavior:** [What you expect to happen]
The program won't compile
**Actual behavior:** [What actually happens]
Giving you a issue
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-07-30:
I can confirm this issue with VSCode as well: [pastebin log](https://pastebin.com/f7iU2FDK).

Ludy87, 2019-07-30:
Try it here, you can not test it yourself.

`\Marlin\src\gcode\config\M575.cpp`

```c++
      if (set0) { MYSERIAL0.end(); MYSERIAL0.begin(baud); }

      #if NUM_SERIAL > 1
        if (set1) { MYSERIAL1.end(); MYSERIAL1.begin(baud); }
      #endif
```

to

```c++
      #if NUM_SERIAL > 1
        if (set1) { MYSERIAL1.end(); MYSERIAL1.begin(baud); }
      #else
        if (set0) { MYSERIAL0.end(); MYSERIAL0.begin(baud); }
      #endif
```

thisiskeithb, 2019-07-30:
Same error:
```c++
Marlin/src/gcode/config/M575.cpp: In static member function 'static void GcodeSuite::M575()':
Marlin/src/gcode/config/M575.cpp:64:31: error: 'class CDCSerial' has no member named 'end'; did you mean 'read'?
         if (set1) { MYSERIAL1.end(); MYSERIAL1.begin(baud); }
                               ^~~
                               read
```

RudyBenner, 2019-07-30:
[image]

shitcreek, 2019-08-07:
I have no issue compiling with BAUD_RATE_GCODE enabled in VSCode and Arduino and without @Ludy87 's change

thisiskeithb, 2019-08-07:
> I have no issue compiling with BAUD_RATE_GCODE enabled in VSCode and Arduino and without @Ludy87 's change

You sure?

Using code I pulled down yesterday, I am still getting:

```c++
Marlin/src/gcode/config/M575.cpp: In static member function 'static void GcodeSuite::M575()':
Marlin/src/gcode/config/M575.cpp:66:31: error: 'class CDCSerial' has no member named 'end'; did you mean 'read'?
         if (set1) { MYSERIAL1.end(); MYSERIAL1.begin(baud); }
                               ^~~
                               read
```

I'll pull down the latest commits from a few minutes ago and rebase my fork and re-test.

thisiskeithb, 2019-08-07:
Using the [latest commit](https://github.com/MarlinFirmware/Marlin/tree/535eeace5f6e68f664169322cacf27b46517d37c), I'm still getting the same error message.

shitcreek, 2019-08-07:
That's strange, I'm using the latest commit as well.

thisiskeithb, 2019-08-07:
It's probably some unique combination of enabled settings since I downloaded a vanilla config & it compiles with `BAUD_RATE_GCODE` enabled for `LPC1768` and `megaatmega2560`. If I have time, I'll take [my fork](https://github.com/thisiskeithb/Marlin/tree/Ender-3-SKR-1.3-TMC5160-Linear-Advance) and start reverting settings until it compiles to see where/why it's failing.

InsanityAutomation, 2019-08-07:
> 
> 
> That's strange, I'm using the latest commit as well.

This will fail on anything except AVR and DUE. The platforms are missing the extended functions. He is using LPC1768 hence why he gets an error.

thisiskeithb, 2019-08-07:
Since more `M575` work is being done on in PR https://github.com/MarlinFirmware/Marlin/pull/14793, it's probably best to wait to see if those changes fix these issues.

InsanityAutomation, 2019-08-07:
@thisiskeithb Itll help for some platforms to be sure. Others will have to wait for platform support. P3P is aware of the change requested for lpc. Not sure if anyone is looking at libmaple / stm32 stuff yet. Time will tell. For now I have sanity checks for supported platforms in that PR. Ill be busy for a bit but ill get back to it soon enough.

Erokawa, 2019-09-24:
Marlin\src\gcode\config\M575.cpp: In static member function 'static void GcodeSuite::M575()':
Marlin\src\gcode\config\M575.cpp:63:29: error: 'class CDCSerial' has no member named 'end'; did you mean 'read'?
       if (set0) { MYSERIAL0.end(); MYSERIAL0.begin(baud); }
                             ^~~
                             read
*** [.pio\build\LPC1768\src\src\gcode\config\M575.cpp.o] Error 1

InsanityAutomation, 2019-09-24:
PR fixing some is still sitting open. Its dead last on my list to work on as use cases for this are incredibly limited. To have a complete working implementation quite a bit is left to do. This will be open for awhile.

boelle, 2019-10-12:
maybe link this issue in the PR? makes it tad more easy to see that a fix has been provided and we just need to wait for the merge... or is patience the word?

InsanityAutomation, 2019-10-15:
#14793

boelle, 2019-12-10:
@Erokawa have you tried to compile with latest copy of marlin 2.0?

compiles fine on arduino ide 1.8.10 

[image]

will try on vs code + platformio extension also

InsanityAutomation, 2019-12-10:
Doesnt matter, its current implementation is fundamentally broken for most platforms. A PR outlining what needs to be done is sitting, but given the extremely small potential user base and pile of other projects itll be awhile before I get back to it.

boelle, 2019-12-10:
yeah i know about the PR... just figured if it would compile at all

gloomyandy, 2019-12-15:
It is also worth noting that changing the baud rate on a CDCSerial connection actually does nothing at all, the connectlon will always operate at full USB speed. 

Oh and if someone is working on the various serial implementations, take a close look at exactly what the flush call does in each case. Unfortunately the Arduino implementation changed from version < 1 to version >= 1. The docs state:
```
Waits for the transmission of outgoing serial data to complete. (Prior to Arduino 1.0, this instead removed any buffered incoming serial data.)
```
Unfortunately the Marlin AVR HAL implementation of MarlinSerial does things the old way (as does the LPC176x Hal and framework), but I'm not sure if all HALs/frameworks will do things the same way.

thinkyhead, 2019-12-16:
A sanity check has been added to catch the option on unsupported platforms.

closed 2019-12-16 by thinkyhead