MarlinFirmware/Marlin #14783: Missing define BLTOUCH_V3

DarkAlchy, 2019-07-30:
I have people telling me to uncomment that line in configuration_adv.h but in last week's cimmit or today's commit there is no sign of this in V2 bugfix.  Any ideas?

thisiskeithb, 2019-07-30:
They are using `bugfix-1.1.x` or haven't updated their configs in a while.

Ludy87, 2019-07-30:
https://github.com/MarlinFirmware/Marlin/blob/c1b99c43b6e3aec37dab84cd34357337f2c845fb/Marlin/src/inc/SanityCheck.h#L214

`BLTOUCH_V3 is obsolete. Please update your configuration.`

Ludy87, 2019-07-30:
https://github.com/MarlinFirmware/Marlin/blob/c1b99c43b6e3aec37dab84cd34357337f2c845fb/Marlin/Configuration.h#L853

https://github.com/MarlinFirmware/Marlin/blob/c1b99c43b6e3aec37dab84cd34357337f2c845fb/Marlin/Configuration_adv.h#L530-L595

DarkAlchy, 2019-07-30:
Thank you everyone.

closed 2019-07-30 by DarkAlchy