MarlinFirmware/Marlin #14784: [FR] [bugfix-2.0.x] revert rotary encoder to validate

spilz87, 2019-07-30:
2 weeks ago, I did a merge, there was something new with rotary encoder : I should turn in the opposite side to validate when I selected a file. 
I was surprised, but finally I like it, I stop validate by mistake. 
But I just did a new merge and it was gone, normal use of the rotary encoder even to validate. 

I don’t know if it was a bug 

My request : 
A #define to select the way when validate 
(Sorry if it already existe, i didn’t find it)

Thanks

robbycandra, 2019-08-01:
Try to uncomment this at configuration.h
//#define REVERSE_SELECT_DIRECTION

spilz87, 2019-08-01:
Uncomment this line will change the direction of the rotation all the time, I want to change it only for confirmation screen

robbycandra, 2019-08-02:
@spilz87 , really ? Please recheck, there are 3 option for Encoder Direction now.

spilz87, 2019-08-02:
My mistake, I was working on an other issue and didn’t see these new options
But after test they are exactly not what I’m looking for : 

- They differentiate menu navigation / value editing like move X 10mm 

- I just want to reverse for validation menu to avoid mechanical validation 
 
It was like this after Scott Lahteine push 3cd9a92dcc525e
But I guess it was not volontaire as it was remove later

robbycandra, 2019-08-03:
I cannot understand why it not suitable with what your looking for.

This topic already have long and a lot of discussion. I think current version will good for everyone if they choose the right option. 

Maybe you should make some table to explain more detail.

spilz87, 2019-08-03:
Value editing -> exemple : move X by 10mm
!=
Validation screen -> exemple : confirm you want to print this file 

It’s not suitable because with actual parameters there are one direction for menu navigation and one direction for value editing. 

What I would like is one direction for menu navigation and value editing and one direction for validation screen.

For me it’s natural to turn clockwise to go down in menu and to increase a value. 
But when there is a choice to confirm like “cancel / confirm” I prefer to change the direction to force to read well the question before clicking 

Is it more clear ? 
I don’t know how to explain it more

spilz87, 2019-08-03:
It was done by Scott Lahteine in push 3cd9a92

spilz87, 2019-08-04:
Actually I do it like this : 

in configuration.h 
> #define REVERSE_VALIDATION_DIRECTION

in menu.cpp 
> bool MarlinUI::selection; // = false
> bool MarlinUI::update_selection() {
>   if (encoderPosition) {
>     // SPILZ87 reverse direction by changing compare from > to <
>     #if ENABLED(REVERSE_VALIDATION_DIRECTION) 
>       selection = int16_t(encoderPosition) < 0;
>     #else
>       selection = int16_t(encoderPosition) > 0;
>     #endif 
>     encoderPosition = 0;
>   }
>   return selection;
> }

it looks like not a lot of people want it, I think we can close.

closed 2019-08-04 by spilz87