MarlinFirmware/Marlin #14786: [BUG] TMC2130 refuse to enable on SKR v1.3 (SPI works)

comps, 2019-07-31:
### Description

Sorry for yet another TMC2130 issue, but I'm stumped. Whenever I try `M17` or a simple `G1`, nothing happens - the motor doesn't move (isn't powered), my lab PSU says ~0A drawn.

A quick rundown:
 * The SKR V1.3 board works fine with A4988, motor enables and moves as expected
 * The drivers also work fine on a RAMPS setup (5V arduino, 12V PSU)
 * I tested genuine waterott drivers, fysetc ones, bigtreetech ones, all TMC2130 v1.0
 * SPI seems to work, see below
 * The NC pin is physically snipped off so it doesn't connect to SLEEP on the board
 * Tried one driver at a time to rule out competition for the MISO line
 * Tried with/without motor connected
 * Using as basic setup as possible (no `SENSORLESS_HOMING`, `SQUARE_WAVE_STEPPING`, etc., just defaults).
 * (Tried both with DIAG1 connected / disconnected.)
 * Using `TMC_USE_SW_SPI` as required for the SKR v1.3
 * Tried both `SOFTWARE_DRIVER_ENABLE` enabled/disabled
 * Measured the pins on the pololu layout and they check out - when enabled, EN is pulled down, when disabled, it's pulled up
 * Tried both 12V and 24V
 * I'm of course always connecting VM after VIO (using 5V from USB via jumper, sits at a comfortable ~3.319V).
 * Tried both `TMCStepper@<1.0.0` and `TMCStepper` in `platformio.ini`

At this point, I'm wondering how well are the V1.0 drivers compatible with 3.3V logic, because this exact thing happened on my breadboard STM32F103 setup - SPI would connect just fine, but the driver wouldn't enable.

Any hints please?

SPI:
```
T:-15>          X       Y       E        
Enabled         false   false   false
Set current     800     800     800
RMS current     1436    1436    1436
MAX current     2025    2025    2025
Run current     25/31   25/31   25/31
Hold current    12/31   12/31   12/31
CS actual               0/31    0/31    0/31
PWM scale       0       0       0
vsense          0=.325  0=.325  0=.325
stealthChop     false   false   false
msteps          256     256     256
tstep>          max     max     max
pwm            
threshold      
[mm/s]         
OT prewarn      false   false   false
OT prewarn has 
been triggered  false   false   false
off time                0       0       0
blank time      16      16      16
hysteresis     
-end5>          -3      -3      -3 
-start          1       1       1  
Stallguard thrs 3       3       0
DRVSTATUS       X       Y       E
stallguard     
sg_result               0       0       0
fsactive       
stst5>          *       *       *    
olb            
ola            
s2gb           
s2ga           
otpw           
ot             
Driver registers:
T:-15>          X       0x80:00:00:00
T:-15>          Y       0x80:00:00:00
T:-15>          E       0x80:00:00:00
Testing X connection... OK
Testing Y connection... OK
Testing E connection... OK
```
(from when I was testing with `SENSORLESS_HOMING`, hence the stallguard thresholds)
(also excuse the `T:-15` printouts from `pronsole`)

Using latest `bugfix-2.0.x` fdbc7333761e1 .

Thanks!

#### Additional Information

* [configs.zip](https://github.com/MarlinFirmware/Marlin/files/3449270/configs.zip)

teemuatlut, 2019-07-31:
You're motors are not moving because in your M122 `off time = 0`.
Only way this should happen is if you've used `SOFTWARE_DRIVER_ENABLE`.

igorepi, 2019-07-31:
> I'm of course always connecting VM after VIO

But you should connect VM before VIO, shouldn't you ? I lost one of my 2130 connecting VM after VIO, so it's really critical.

From [Watterott FAQ](https://learn.watterott.com/silentstepstick/faq/):

> At power-up the motor supply voltage VM should come up first and then the logic supply voltage VIO.

comps, 2019-07-31:
> You're motors are not moving because in your M122 `off time = 0`.
> Only way this should happen is if you've used `SOFTWARE_DRIVER_ENABLE`.

It's possible that I took the `M122` printout while I was still experimenting with various options to make it work, but I've also tried it without `SOFTWARE_DRIVER_ENABLE`, as mentioned. I just didn't want to post some 14 `M122` printouts for every combination of options.

> From [Watterott FAQ](https://learn.watterott.com/silentstepstick/faq/):
> 
> > At power-up the motor supply voltage VM should come up first and then the logic supply voltage VIO.

Hmm, I guess that depends on the stepstick used - the page says that for variable 3.3-5V sticks, VM should come up first to power the internal TMC voltage regulator, but for 5V-only sticks (presumably), it *must not*:

> The 5V logic SilentStepSticks do not use the internal voltage regulator of TMCxxxx and therefor only a 5V supply voltage for VIO is possible and **VM has not to be present before VIO**. 

... Regardless, I think I solved the problem by flipping the board power switch from USB 5V to on-board regulator, so it might have been the VM-before-VIO thing even if that doesn't make much sense given that the drivers were clearly responding on SPI and could hold state (ie. stallguard bits).

It's also possible that the TMC drivers consume a LOT more current on VIO when enabled and the USB supply couldn't deliver this with voltage dropping below logic limits for a few ms, don't have an oscilloscope handy to find out.

closed 2019-07-31 by comps